PX4/PX4-Autopilot #22385: Drone isn't stopping on releasing Roll/Pitch stick of RC while using ARKFlow for local positioning

MuhammadBilal1, 2023-11-16:
### Describe the bug

I performed two flights with the [ARKFlow](https://arkelectron.com/product/ark-flow/) sensor in "Position" flight mode.
I was expecting the stable flight as shown in the [video](https://www.youtube.com/shorts/aPQKgUof3Pc) by ARKFlow.
Followed the configurations given on the [PX4 documentation](https://docs.px4.io/v1.14/en/dronecan/ark_flow.html) about ARKFlow.

### Problem:
**Primary Problem:** (the drone isn't stable in position mode)

- Initially the takeoff was not good (the drone was circling).
- Upon releasing the Roll/Pitch stick of RC, the drone wasn't stopping, then I had to move the stick to opposite direction to stop the drone.
- Switching to "Land" mode didn't worked.
- Sometimes, it was holding the position and sometimes it was moving freely while hovering.

**Secondary Problem:** (the parameters not found which are given in PX4 documentation)

I wasn't able to find the following parameters in PX4 [1.14.0 (Stable)](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0), whereas, the [PX4 documentation](https://docs.px4.io/v1.14/en/advanced_config/parameter_reference.html#:~:text=SENS_FLOW_MAXHGT%20(FLOAT)) says the parameters are available and the parameters are available in the PX4 [source code](https://github.com/PX4/PX4-Autopilot/blob/v1.14.0/src/modules/sensors/sensor_params_flow.c) also.
- SENS_FLOW_MINHGT
- SENS_FLOW_MAXHGT
- SENS_FLOW_MAXR

### Tried the same ARK Flow sensor on two different autopilots:
1. [Pixhawk 6C](https://holybro.com/products/pixhawk-6c)
2. [ARKV6X](https://arkelectron.com/product/arkv6x-bundle/)

### Tried solutions to find the missing parameters:
1. Clear the cache of QGroundControl from "appdata" directory of windows 11.
2. Uninstalled (from ControlPanel) and installed the latest version of QGroundControl (v4.2.9)

### To Reproduce

Perform flight with ARKFlow in position flight mode without GPS.

### Expected behavior

Drone should fly stable in local positioning without GPS, and using ARKFlow sensor.

### Screenshot / Media

`param show UAVCAN_RNG_MIN` :arrow_down: 

[image]

`uavcan status` :arrow_down: 
[image]

[image]

### Flight Log

Flight Log (PX4 - Pixhawk 6C):
https://review.px4.io/plot_app?log=d0797dde-2bf8-41be-8ae3-b0104ebedf24

Flight Log (PX4 - ARKV6X)
https://review.px4.io/plot_app?log=2644a76b-d062-4220-b4f1-c22c4661b438

### Software Version

PX4 [1.14.0 ](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0)(Stable)

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

AlexKlimaj, 2023-11-16:
In your ARKV6X log, EKF2_OF_CTRL is set to 0, which caused position mode not to activate. It was in altitude mode the whole flight. Also UAVCAN_ENABLE is set to 0 instead of 2. 

The params are all in the 6C log.
https://review.px4.io/download?log=d0797dde-2bf8-41be-8ae3-b0104ebedf24&type=1

But they are missing in the ARKV6X log due to UAVCAN_ENABLE being set to 0.
https://review.px4.io/download?log=2644a76b-d062-4220-b4f1-c22c4661b438&type=1

MuhammadBilal1, 2023-11-16:
> In your ARKV6X log, EKF2_OF_CTRL is set to 0, which caused position mode not to activate. It was in altitude mode the whole flight. Also UAVCAN_ENABLE is set to 0 instead of 2.
>
> The params are all [...]

Ok. **What's the feedback on the Pixhawk 6C?**
It's my bad I might have missed the parameters, but what about the flight with the 6C, whereas, all the required parameters were there.

AlexKlimaj, 2023-11-16:
I would recommend building and flashing main on the ARK Flow and testing again. 

The default position mode is smooth acceleration, so it won't stop immediately while it slows down.

MuhammadBilal1, 2023-11-17:
Ok. I will try to flash the ARK Flow using the instructions [here](https://docs.px4.io/main/en/dronecan/px4_cannode_fw.html#building-the-firmware).

> The default position mode is smooth acceleration, so it won't stop immediately while it slows down.

PX4 documentation says otherwise about the [Position](https://docs.px4.io/main/en/getting_started/flight_modes.html#position-mode-mc:~:text=When%20the%20sticks%20are%20released/centered%20the%20vehicle%20will%20actively%20brake%2C%20level%2C%20and%20be%20locked%20to%20a%20position%20in%203D%20space) flight mode.

What should be the solution if I want **"When the sticks are released/centered the vehicle will actively brake, level, and be locked to a position in 3D space"** (while using ARK Flow and without the use of GPS)?

AlexKlimaj, 2023-11-17:
> Ok. I will try to flash the ARK Flow using the instructions [here](https://docs.px4.io/main/en/dronecan/px4_cannode_fw.html#building-the-firmware).
>
>
> PX4 documentation says otherwise about the [...]

Yes it does a smooth acceleration brake until it stops and holds position. Can you share a video of the flight behavior?

MuhammadBilal1, 2023-11-17:
> Yes it does a smooth acceleration brake until it stops and holds position. Can you share a video of the flight behavior?

Here is the video of the flight: https://vimeo.com/885640985
Password: pass654123

Log: https://review.px4.io/download?log=d0797dde-2bf8-41be-8ae3-b0104ebedf24&type=1

AlexKlimaj, 2023-11-17:
It could be that floor is too difficult for the optical flow sensor, reflective with few features. You could try placing some cross hatch tape on the floor as a test.