PX4/PX4-Autopilot #22409: Battery Hotswap -- SoC and mAh reset

dakejahl, 2023-11-19:
### Describe problem solved by the proposed feature

A user wants to hotswap batteries when there are multiple connected in parallel and have the SoC and mAh consumed reset. 

### Describe your preferred solution

Allow the `Battery` class to detect the "disconnected" state based on current. Add new parameters BAT${i}_HTSWP_EN BAT${i}_I_HTSWP_DT to enable the hotswap detection feature and set the absolute value current threshold for detecting that a battery is disconnected. This allows the user to adjust the current threshold depending on the idle current draw of their system.

### Describe possible alternatives

Add a button in QGC and a mavlink command to allow the user to manually reset the SoC and mAh consumed per battery.

### Additional context

The `Battery` class already uses a minimum voltage to set the "connected" state, but for parallel batteries we need to use current.

dakejahl, 2023-11-22:
This is what a hotswap looks like on a system with two 12S batteries connected in parallel. The batteries start off about halfway discharged and are replaced with fully charged batteries.

[image]

sfuhrer, 2023-12-15:
@MaEtUgR fyi