PX4/PX4-Autopilot #22415: [Bug] The drone starts rotating by himself when in Hold mode

LukyKozlojc, 2023-11-20:
### Describe the bug

Hello,

I started experiencing strange behavior of my drone. I am running the px4 via "make px4_sitl gz_x500" and then when I swtich the drone via QGroundControl to takeoff, everything is working correctly, but after a few seconds in hold mode he starts rotating on the left side and then on the right side by small amount of degree, but the degree is increasing with time. It is happening also when I run offboard mode and I am controling the drone's tilt via ros2 LQR controller, so yes, I am sending the keep alive message into mavlink for the offboard mode.
I do not think that this could be happening only for me, so thats why I am reporting this as a bug.
Or if you have any suggestions or solutions, I would be glad.

Thank you.

### To Reproduce

1. run make px4_sitl gz_x500
2. run QGroundControl
3. press takeoff button in QGroundControl
4. wait a few seconds and spectate the drone's orientation.

### Expected behavior

I believe that the drone should be holding his position without any changes in orientation.

### Flight Log

I do not have any, sorry.

### Software Version

latest to this date 20.11.2023

### Flight controller

I am using only simulation yet

### Vehicle type

Multicopter

dakejahl, 2023-11-21:
please share a log, logs from sim can be found at **PX4-Autopilot/build/px4_sitl_default/rootfs/log/**

LukyKozlojc, 2023-11-22:
I have found out, that the issue was caused by my mistake, so I am closing the issue

closed 2023-11-22 by LukyKozlojc