PX4/PX4-Autopilot #22417: Sensor priorities not used for EKF selection

julianoes, 2023-11-20:
### Describe problem solved by the proposed feature

From what I can see, the sensor priorities as set by the params `CAL_ACCx_PRIO` (and others for other sensors) are not used in the EKF selection.

### Describe your preferred solution

Should we publish the sensor priority as part of the `sensor_accel` (and others) topics and then use them in the EKFSelector in the case when the "test ratio" is the same between multiple EKFs?

@dagar and @bresch do you have opinions?

### Describe possible alternatives

Or should we remove these priorities altogether?

### Additional context

FYI @vincentpoont2

bresch, 2023-11-20:
Related to https://github.com/PX4/PX4-Autopilot/pull/21815

AlexKlimaj, 2023-11-21:
We should keep the priorities. Publishing them as part of the sensor messages could clean up the logic. Ideally the multi ekf takes into account the imu and mag priorities when switching instances.

antonio-sc66, 2023-12-05:
Hi, @julianoes, that feature seems to work well but hasn't been integrated, not sure if they want me to explain anything about it. I will try to get it merged