PX4/PX4-Autopilot #22428: [Bug] The Breach Return Point set in QGC is invalid.

Drone-Lab, 2023-11-22:
### Describe the bug

When we set up a geofence and a Breach Return Point in QGC, and manually fly the drone into the geofenced area, even with RTL TYPE set to 3 (Closest Safe Destination Return Type, RTL_TYPE=3), the drone ignores the Breach Return Point and returns to the home point instead. It appears that the Breach Return Point is a non-functional feature. I couldn't find any documentation for this feature in the official PX4 documentation

### To Reproduce

1. Take off the drone and upload the geofence. Add a Breach Return Point near the geofence in QGC.
[image]

2. Manually control the aircraft to fly towards the restricted area, making contact with the geofence (It's not possible to set a direct course into the restricted area, but you can set a route outside the restricted zone).
[image]

### Expected behavior

Even after making contact, the drone still does not fly towards the Breach Return Point.
[image]

### Flight Log

none

### Software Version

HW arch: PX4_SITL
PX4 git-hash: 83841d1ad183e1b874fad1d07e5cd6a448d0c90e
PX4 version: 1.14.0 0 (17694720)
PX4 git-branch: main
OS: Linux
OS version: Release 5.15.0 (84869375)
Build datetime: Nov 14 2023 17:45:03
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.4.0
PX4GUID: 1006554d49534954414c44494e4f30303030
UNKNOWN MCU

### Flight controller

none

### Vehicle type

None

sfuhrer, 2023-12-15:
With current main it works as expected. I here set RTL_TYPe to 3.
https://github.com/PX4/PX4-Autopilot/assets/26798987/61167af5-4649-45a8-b367-b2561b22ca85

Drone-Lab, 2023-12-15:
> With current main it works as expected. I here set RTL_TYPe to 3. https://github.com/PX4/PX4-Autopilot/assets/26798987/61167af5-4649-45a8-b367-b2561b22ca85

Nice work,thanks for your contribution!

closed 2023-12-15 by Drone-Lab