PX4/PX4-Autopilot #22429: [Bug] ninja: error: unknown target 'ignition'

LShivaRudra, 2023-11-22:
### Describe the bug

I'm trying to simulate a quadcopter in sitl and when I try to build the PX4 software with simulation using:
`make px4_sitl ignition`

I get the error:
`ninja: error: unknown target 'ignition'`
`make: *** [Makefile:232: px4_sitl] Error 1`

The same error occurs when I use: 
`make px4_sitl gz` or
`make px4_sitl gz_x500`

### To Reproduce

I used the following configurations for the softwares:
Ubuntu Version: 20.04
PX4-Autopilot: v1.14.0
ROS version: ros2 foxy
Ignition version installed: citadel

### Expected behavior

-

### Flight Log

-

### Software Version

v1.14.0

### Flight controller

I'm trying to simulate(sitl)

### Vehicle type

Multicopter

avcuenes, 2023-12-09:
If you use ubunut 20.04 , you have to insatall gazebo ignition manually 
Lool this documantation
`https://docs.px4.io/v1.14/en/sim_gazebo_gz/`

LShivaRudra, 2023-12-09:
Okay, I'll have a look at it and get back to you. Thanks!

closed 2023-12-11 by LShivaRudra