PX4/PX4-Autopilot #22430: [Bug] Matek H743-SLIM V3 only MAIN 1 and MAIN 2 output PWM signal

dollop80, 2023-11-22:
### Describe the bug

After a successful building from sources and flashing firmware for the Matek H743-SLIM V3 board I tried to perform the actuator testing from the Actuators tab. Only two motors were working, configured for MAIN 1 and MAIN 2 (I use 100% working 4in1 ESC). I checked the MAIN outputs for signals with scope - the MAIN 1 and MAIN 2 had signals, MAIN 3 and MAIN 4 output was logical 0. I checked with all protocols (Dshot and PWM) with the same result

### To Reproduce

 1. build and flash latest firmware
 2. configuring the board for generic quad
 3. calibrate the sensors and RC 
 4. try to test motors from Actuators tab

### Expected behavior

As I move a motor test slider a linked motor should rotate. It is tur only for the two outputs - MAIN 1 and MAIN 2

### Flight Log

not relevant

### Software Version

v1.15.0dev

### Flight controller

Matek H743-SLIM V3

### Vehicle type

Multicopter

dollop80, 2023-11-24:
Setting the parameter **UAVCAN_ENABLE=disable** (enabled by default) fixes the problem with MAIN 3-6 group PWM output. 
So, if you need MAIN 3-6 group PWM output, you can not use UAVCAN. But it is another issue ;)

closed 2023-11-24 by dollop80

avinashkrg90, 2026-02-16:
Is there any way to keep the UAVCAN enabled and get PWM outputs in the motor group pins 3-6. Its basically the timer issue where same timer is being shared by both the hardwares. Can we change the timer used by any of the hardwares, so that both features are useful at the same time.
Thanks