PX4/PX4-Autopilot #22431: support hardware simulation mode for gz_x500 of gazebo garden

zyy37, 2023-11-22:
### Describe problem solved by the proposed feature

Please support hardware simulation mode for `gz_x500` of gazebo garden. Currently, [HITL](https://docs.px4.io/main/en/simulation/hitl.html#gazebo-classic) only supports gazebo-classic, and [SIH](https://docs.px4.io/main/en/sim_sih/#setting-up-the-display) only supports jMAVSim. Isn't gazebo garden recommended? So I hope HITL or SIH supports gz_x500, rather than switching between various simulations. Thanks.

### Describe your preferred solution

n/a

### Describe possible alternatives

n/a

frede791, 2023-11-24:
@zyy37 I believe the gazebo garden x500 is supported in sihsim under the name quadx. 
You can call it up using `make px4_sitl sihsim_quadx`.

federicociresola, 2024-07-30:
I am interesting in this feature too. Is there some news on SIH with Gazebo Garden/Harmonic?