PX4/PX4-Autopilot #22446: I2C sensor address changed to default on Pixhawk restart

MuhammadBilal1, 2023-11-27:
### Describe the bug

I am connecting _two_ [LIDAR-Lite V3](https://www.garmin.com/en-US/p/557294) sensors on the I2C port of Pixhawk 6C using the [I2C Splitter](https://www.amazon.com/Pixhawk-i2c-Splitter-Pixhawk-Controller-Hobbypower/dp/B00LAWOUE4).

### Purpose of connecting two Range Finders:
1- First [range finder](https://docs.px4.io/main/en/sensor/lidar_lite.html) will be used as distance sensor (facing downward)
2- Second range finder will be used as [Collision Prevention](https://docs.px4.io/v1.14/en/computer_vision/collision_prevention.html) (facing forward)

### Drone Setup:
Autopilot: Pixhawk [6C](https://holybro.com/products/pixhawk-6c?variant=42420293402813)
Firmware: PX4 [1.14.0](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0)
GPS: None
Local Positioning Sensor: [ARKFlow](https://docs.px4.io/v1.14/en/dronecan/ark_flow.html) (connected at CAN port)
Range Finder: [LIDAR-Lite V3](https://docs.px4.io/main/en/sensor/lidar_lite.html)

### Range Finders Configuration: (Analyze > Mavlink Console)
Starting collision prevention range finder (facing forward) with below command:
nsh> ```ll40ls start -X -a 95 -R 0```

Starting range finder (facing downward) with below command:
nsh> ```ll40ls start -X -a 98```

### What's happening:
Both sensors are connected to the I2C splitter, both are working as fusion (both are giving only the smallest value).

### Problem?
1- Is this possible to use multiple LIDAR-Lite V3 or LIDAR-Lite v3HP sensors as collision prevention (Forward, Rear, Right, Left) sensors?
2- What's the solution to orientation not being saved permanentaly? Upon restarting the orientation, the orientation goes back to the default (25).
3- Is there any other industrial grade sensor which I can use as collision prevention on all 5 sides of my drone?

[image]
 

### What I tried?
I tried to connect one Range Finder on I2C and other on PWM port, but with this approach I am not getting distance sensor data from the sensor connected with PWM.
```ll40ls_pwm start```

### To Reproduce

None.

### Expected behavior

Collision Prevention should work.

### Flight Log

None.

### Software Version

PX4 1.14.0 

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

MuhammadBilal1, 2023-12-13:
@bresch  @bkueng  Can you please help me here or guide me to the one who knows the details of this.

bkueng, 2023-12-18:
> 1- Is this possible to use multiple LIDAR-Lite V3 or LIDAR-Lite v3HP sensors as collision prevention (Forward, Rear, Right, Left) sensors?

Yes but I'm not sure if that has been tested.

> 2- What's the solution to orientation not being saved permanentaly? Upon restarting the orientation, the orientation goes back to the default (25).

You need to add both of your commands to the [startup script](https://github.com/PX4/PX4-Autopilot/blob/main/ROMFS/px4fmu_common/init.d/rc.sensors) (`ll40ls start -X -a 95 -R 0` and `ll40ls start -X -a 98`)