PX4/PX4-Autopilot #22464: Fix and rewrite obstacle avoidance

AlexKlimaj, 2023-11-30:
### Describe problem solved by the proposed feature

Currently obstacle avoidance in PX4 is broken.

### Describe your preferred solution

Move to vector field histogram (VFH) algorithm.

Don't hit obstacles that you can see, apply negative force to push drone away if within minimum distance, only apply allowed components of sticks vector while up against the invisible wall

Handle arbitrary obstacle_distance message formats, ie allow greater than or less than 10 degree angle step sizes, allow more or less than 36 sectors

Work in the default position mode (manual acceleration

Another nice option would be params (reboot required) to adjust number of sectors, so you can have bigger/smaller resolution if you have the memory/cycles for it. Maybe set defaults scaled to each hardware platform.

It should work with distance_sensor and obstacle_distance as uorb inputs.

### Describe possible alternatives

Remove obstacle avoidance from PX4 and expect it to be done on the companion computer in offboard mode.

Jaeyoung-Lim, 2023-12-01:
@AlexKlimaj https://github.com/PX4/PX4-Avoidance was already using VFH, but also probably broken.

Is this concerning collision prevention? or obstacle avoidance? Obstacle avoidance never really existed on the PX4 level and usually hard to achieve with reactive methods in general.

aazam-gh, 2023-12-05:
/attempt #22464

dirksavage88, 2024-04-12:
@OgnjenX the broken part of collision prevention is that the guide angle tuning doesn't work. But also not working with the default MPC_POS_Mode probably doesn't help either.

 Collision prevention should not only stop the vehicle but also give setpoints around the obstacle to "slide" past the obstacle. I believe it was intended as a middle ground between basic obstacle detection and stopping and a full fledged companion computer that incorporates a nav stack/planner.

Sambit003, 2024-05-18:
/attempt #22464

ydv129, 2024-06-23:
/attempt #22464

Claudio-Chies, 2024-07-22:
/attempt #22464

vishwamartur, 2024-11-13:
/attempt #22464

AlexKlimaj, 2024-11-25:
@Claudio-Chies Great work! Let me figure out how to award the bounty.

Claudio-Chies, 2024-12-02:
@AlexKlimaj Thanks alot!

closed 2024-12-02 by Claudio-Chies