PX4/PX4-Autopilot #22465: [Bug] yaw rate setpoint goes out of control

kenta-ishibashi-td, 2023-11-30:
### Describe the bug

After the Octocopter had been flying normally for a while, the drone suddenly made two rotations around the yaw axis. While spinning, it hit a nearby building and nearly crashed.Please see flight review.

### To Reproduce

1. Drone switched on
2. Switched to stabilized mode
3. Took off
4. See error

### Expected behavior

No rotation in Yaw direction

### Screenshot / Media

[image]

### Flight Log

https://review.px4.io/plot_app?log=e79ca622-a5d7-4b76-9d08-26e8b75273e6

### Software Version

・v1.13.3 (pre-build)
Downloaded on QGC

### Flight controller

modalai fc-v1

### Vehicle type

Multicopter

danielhonies, 2023-12-01:
Could this be a mechanical defect (like a loose prop) that occurred before it went out of control?

kenta-ishibashi-td, 2023-12-04:
@danielhonies 
Thank you for your reply.
Before and after the flight, I checked the propeller and motor-mounts for looseness.
And it was no problem....