PX4/PX4-Autopilot #22482: [ARKFlow (No GPS)] Delayed response to RC sticks (Roll/Pitch/Yaw) in Position mode only

MuhammadBilal1, 2023-12-05:
### Describe the bug

# Drone Setup:
**Autopilot:** Pixhawk [6C](https://holybro.com/products/pixhawk-6c?variant=42420293402813)
**Firmware:** PX4 [1.14.0](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0)
**GPS:** None
**Local Positioning Sensor:** [ARKFlow](https://docs.px4.io/v1.14/en/dronecan/ark_flow.html) (connected at CAN port, light of ARKFlow sensor is solid blue)
**Range Finder:** [LIDAR-Lite V3](https://docs.px4.io/main/en/sensor/lidar_lite.html)

Today, I just tuned my new drone frame in “Altitude” mode.
It went well and I can fly freely in Altitude mode.

But, as soon as I started flying in Position mode, the response to RC sticks is very bad. Upon doing the movements from RC sticks, the drone response is delayed it takes almost 1-2 seconds to the response and the response is also not seems stable.

# Problem:
- Why the drone is showing delay is response as soon as I switch to the position flight mode?
- Why the drone becomes shaky and out of control since it's flying fine in Altitude mode?

# My Hunch!
* Is the problem due to the ARKFlow sensor?
* Is the problem due to the PID Tuning?
* Is this something related to any of the following!

**1- [Attitude Controller](https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html#attitude-controller)**
[image]

**2- Velocity Controller**
[image]

**3- [MC Filter Tuning & Control Latency](https://docs.px4.io/main/en/config_mc/filter_tuning.html)**

### To Reproduce

Fly with ARKFlow sensor and without GPS in position flight mode

### Expected behavior

The drone should be stable enough to fly indoor without getting crash.

### Flight Log

**Altitude flight mode log**: (Control is almost fine)
https://review.px4.io/plot_app?log=63a3002b-acdd-448f-b6f1-64aa8f335151

**Position flight mode log**: (delayed response to RC sticks)
https://review.px4.io/plot_app?log=df54291a-da49-439d-90e6-acc6e7f6f850

**Position flight mode log**: (delayed response to RC sticks)
https://review.px4.io/plot_app?log=3b2384e8-8aba-4b11-926c-ddf89fb29799

### Software Version

1.14.0

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

bresch, 2023-12-05:
I see that you're using the "old" position setpoint generator ("smoothed velocity"). The newer one provides a much better stick response and intuitive reaction (much more like altitude mode with position locking). To set it, change [MPC_POS_MODE](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MPC_POS_MODE) to "Acceleration based".

adbidwai, 2024-07-01:
@MuhammadBilal1 I am using a motion capture system and I am encountering similar issues. Did you find any solution to your problem? Let me know! :)
Thanks

bresch, 2024-07-16:
Closing as there is no answer from @MuhammadBilal1 .

closed 2024-07-16 by bresch

MuhammadBilal1, 2024-07-18:
> @MuhammadBilal1 I am using a motion capture system and I am encountering similar issues. Did you find any solution to your problem? Let me know! :) Thanks

I followed the instruction of https://github.com/PX4/PX4-Autopilot/issues/22482#issuecomment-1840342414 and it worked for me.