PX4/PX4-Autopilot #22485: [Bug] PX4 does not compile with instructions from Getting started document

sprobst, 2023-12-05:
### Describe the bug

I followed the instructions on Getting started. I installed all Python dependencies and checked out the latest on main branch. However I receive a fatal error, because one declaration shadows another:
```
In file included from /[...]/PX4-Autopilot/src/modules/ekf2/EKF/yaw_fusion.cpp:34:
In file included from /[...]/PX4-Autopilot/src/modules/ekf2/EKF/ekf.h:66:
/[...]/PX4-Autopilot/src/modules/ekf2/EKF/aux_global_position.hpp:92:3: fatal error: declaration shadows a variable in namespace 'estimator' [-Wshadow]
                HPOS  = (1<<0),
                ^
/[...]/PX4-Autopilot/src/modules/ekf2/EKF/common.h:138:2: note: previous declaration is here
        HPOS  = (1<<0),
        ^
1 error generated.
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

I tried switching to latest Tag but there I get another error. 
The command I use to compile is: `make px4_sitl jmavsim`
I'm on macOS 13.6.1 (22G313)
Apple clang version 15.0.0 (clang-1500.0.40.1)
Target: x86_64-apple-darwin22.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

I also ran the macOS.sh file in Tools directory, because I initially had other compile issues.

### To Reproduce

1. Checkout main branch
2. Run setup scripts
3. Run `make px4_sitl jmavsim`

### Expected behavior

It compiles and jmavsim starts

### Flight Log

no flight log

### Software Version

ve all

### Flight controller

none

### Vehicle type

None

bresch, 2023-12-06:
fixed by https://github.com/PX4/PX4-Autopilot/pull/22484

@sprobst You can now pull the newest changes on `main` an build without issues.

sprobst, 2023-12-06:
I noticed last night. It now works.