PX4/PX4-Autopilot #22490: Support for logging via NAND flash

j-chen-opteran, 2023-12-05:
### Describe problem solved by the proposed feature

- Boards such as the KakuteH7 V2/H7 Mini have a Winbond 25N NAND flash chip instead of a SD card slot
- The small size of the H7 Mini as well as the beefy H743 chip make it well suited for visual odometry and SLAM experimentation on micro-sized aircraft
- PX4 currently has no provision to log flight data on these boards at all. Doing some digging into the codebase shows that IO to a SD card is handled by NuttX, and NuttX doesn't seem to have an easy way to interface with NAND flash memory.

### Describe your preferred solution

- Not entirely sure of PX4's relationship with NuttX, but to do it elegantly I assume the RTOS needs a bit of work to have NAND memory IO implemented
- This would be clean and separate duties between the flight control software and the RTOS, but unfortunately I lack the expertise to start working on this 

### Describe possible alternatives

Ardupilot contains a driver that uses the NAND flash chip over SPI:
https://github.com/ArduPilot/ardupilot/blob/c5b9f5051f997ec3f0f223380f912f5f51de641f/libraries/AP_Logger/AP_Logger_W25N01GV.cpp#L8
Something like this could be feasible.

julianoes, 2023-12-05:
First step would be to add it to NuttX, I think.

Right @davids5?

davids5, 2023-12-06:
@j-chen-opteran have a look at https://lists.apache.org/thread/kh5t6fn24q97d3qxzthksld4l3jtw27c and see if this offers any indication it has been  worked on, or added.

j-chen-opteran, 2023-12-06:
Thanks for the quick replies! Did some digging in the direction @davids5 pointed. It appears NuttX itself has [partial support for NAND flash](https://nuttx.apache.org/docs/latest/components/drivers/special/mtd.html) and I'm happy to have a crack at implementing this. 

However I see that PX4 submodules NuttX - this is a bit new to me and I'd like to start the right way. I assume code changes has to be done in the [NuttX repo](https://github.com/PX4/NuttX/tree/3dc3cf522758d70ca2a07c156e5e02517650327d) first?

Thanks.

iftahnaf, 2024-01-22:
Hi @j-chen-opteran, are you working on this? I can try to help. I don't have a lot of experience in that field, but I'm eager to learn.

sstroobants, 2024-07-04:
Is anyone working on this? If I understand correctly, V1.3 of the KakuteH7 mini even has 1GB of NAND Flash, would be a shame to let that go to waste.

julianoes, 2024-07-07:
@sstroobants contributions welcome.

sstroobants, 2024-07-09:
I would love to, but I'm afraid I lack the skills to develop something as embedded as this. If I find time I might try to see what is possible, but a first glance in the NuttX documentation makes me believe it has to be built from the ground up.

sstroobants, 2024-10-01:
Again, I'd love for this to work and would want to spend some time on this. I do lack the skills to start on this however so if there is anyone that wants to help me with this, that'd be much appreciated.

dagar, 2024-10-01:
@sstroobants it depends what actual hardware you're talking about, but potentially everything you need should be in place as of https://github.com/PX4/PX4-Autopilot/pull/23424.

sstroobants, 2024-10-02:
@dagar thanks for the response. I am not exactly sure what you mean. The problem (I think), is support for NAND flash in [NuttX](https://nuttx.apache.org/docs/latest/components/drivers/special/mtd.html) and [LittleFS](https://github.com/littlefs-project/littlefs/issues/11) which are both not (fully) implemented. Again, my knowledge on these low-level things is very limited so correct me if I'm wrong. 
From what I understand, there needs to be a driver for the W25N01GV and also support on NuttX side to do this nicely.

alnI3S, 2025-09-29:
Hi,
I'm working on this Holybro Kakute H7 Mini with Winbond W25N01GV Nand Flash QuadSPI.
 I do write the mtd driver for this w25n01gv chip but I'm stuck on initialize it using qspi bus in the board init.c.
I also see that some board using qspi (like spracing/h7extreme board) "copy" the stm32_qspi.* from NuttX submodule to PX4 board specific folder, that's not nice. So for now I'm trying to "include" the nuttx/arch/arm/src/stm32h7/stm32_qspi.h in my init.c without success.
So I need help about importing NuttX drivers.