PX4/PX4-Autopilot #22492: [Bug] Drone does not stop when clicking Pause in QGC during mission mode

Drone-Lab, 2023-12-06:
### Describe the bug

We have identified a bug where, during the execution of a mission, clicking 'Pause' in QGroundControl (QGC) does not cause the drone to stop as per the QGC manual. Instead, the drone completes the current task and flies to the next waypoint before stopping. Additionally, attempting to 'Hold' again during this process does not halt the drone.

We identified the issue in the 

> PX4-Autopilot-main\src\modules\navigator\loiter.cpp file.

 In the void Loiter::on_activation() function, the problem arises because the if branch is executed instead of the else branch. 
[image]

### To Reproduce

1. Plot the mission.
2. Drone executes the mission.
3. While executing the mission (not in takeoff or ascent/descent states), click 'Pause' in QGroundControl (QGC).

### Expected behavior

The drone does not stop immediately as per the manual; instead, it continues until reaching the mission waypoint before coming to a halt.

### Screenshot / Media

[image]
[image]

### Flight Log

[issue_mission_pause_fail.zip](https://github.com/PX4/PX4-Autopilot/files/13579083/issue_mission_pause_fail.zip)

### Software Version

HW arch: PX4_SITL
PX4 git-hash: 9d465615d11ab4b570a66da22a3681f56679a317
PX4 version: 1.15.0 40 (17760320)
PX4 git-branch: main
OS: Linux
OS version: Release 5.15.0 (84869375)
Build datetime: Nov 30 2023 15:12:54
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.4.0
PX4GUID: 1006554d49534954414c44494e4f30303030
UNKNOWN MCU

### Flight controller

SITL

### Vehicle type

None

ryanjAA, 2023-12-09:
Same. Confirmed. If you hit hold it will stop but pause is broken and does continue.

Drone-Lab, 2023-12-09:
> Same. Confirmed. If you hit hold it will stop but pause is broken and does continue.

Yes.It's not a ui bug, the drone is indeed in loiter/hold mode.But it seems to only execute part of the code in hold mode.

sfuhrer, 2023-12-15:
I'm not able to reproduce, for me it always stops when I pause. This is with current main and QGC 4.2.8
https://github.com/PX4/PX4-Autopilot/assets/26798987/e2d51215-3a1a-482a-82a0-c080e248201b

Drone-Lab, 2023-12-15:
Thanks ,I reproduce it on  QGC 4.2.8 and current main agaian. 

When you try to reproduce this bug, let the drone fly for a while (for example, between waypoint 2 and 3) and wait until it is highly stabilized before hitting pause.

We've pinpointed the cause of the problem, which is due to a multi-threaded race condition, so this doesn't always trigger (but most do). Additionally this issue doesn't trigger when the drone takes off, as pause while taking off, will use the current position as the mission target.

ryanjAA, 2023-12-16:
I’ll make a video. Does it every-time in sitl for me.

Drone-Lab, 2023-12-18:
https://github.com/PX4/PX4-Autopilot/assets/151698793/93dfb37a-6255-4b8d-86b4-cafedab6c7a8

This is with current main.

sfuhrer, 2023-12-18:
> can.t.pause.mp4
> 
> This is with current main.

Thanks for reproducing again and sharing the video. Do you per chance still have the log file? You can find it for SITL flights under `/build/px4_sitl_default/rootfs/log`

closed 2024-01-22 by dagar