PX4/PX4-Autopilot #22512: [Bug] intereference between Position and Offboard mode

ayhamalharbat, 2023-12-08:
### Describe the bug

When in Position mode and there is an offboard controller publishing `vehichle_attituide_setpoint` commands, you notice interence between the offboard controller commands and the PX4 controllers commands.

Notices in SITL and in Hardware experiments

### To Reproduce

1. Try to run this [demo](https://github.com/SaxionMechatronics/px4_offboard_lowlevel/blob/main/instructions/demo.md)
2. Remember to switch to offboard mode, and arm the drone
3. The drone will follow a circle trajectory
4. switch to position mode when the offboard controller is still active.

### Expected behavior

Motor 'twiching', and unstable behaviour

### Flight Log

SITL 1: https://review.px4.io/plot_app?log=202814b8-1fed-4209-b967-1f6dd080f1f3
SITL 2: https://review.px4.io/plot_app?log=04aab48e-2276-4aab-acbb-6b5f99df64cb
Hardware: https://review.px4.io/plot_app?log=2d09158b-d315-4b4f-8237-b0df13b600fd

### Software Version

Main @commit e35380d

### Flight controller

Holybro Pixhawk 4

### Vehicle type

Multicopter

Jaeyoung-Lim, 2023-12-08:
@ayhamalharbat Unfortunately I cannot reproduce this problem

Offboard control with attitude setpoint seems to work as expected: https://review.px4.io/plot_app?log=e34f52f5-fb65-44af-9f42-780668782a58

Are you sure that the choppy attitude setpoints are not coming from the offboard process?

ayhamalharbat, 2023-12-09:
> Offboard control with attitude setpoint seems to work as expected on latest `main`

@Jaeyoung-Lim  maybe I wasn’t clear:
- Offboard is working
- Switching to Position mode while the offboard controller is still running (as in: still publishing its commands on the ros2 topic `/fmu/in/vehicle_attitude_setpoint)` does not work
- The problem is visible in Position mode

I tried it with PX4-Manipulation:
- You will have the same problem if you remove this check:
https://github.com/Jaeyoung-Lim/px4-manipulation/blob/97a8a8f37cfe5f9cd861c2223ebb093c4038d4c3/px4_manipulation/src/px4_manipulation.cpp#L97C16-L97C16
- Flight log: https://review.px4.io/plot_app?log=761a9021-f03a-49f8-b70c-283cbea0f618

For safety reasons, I think that the ROS 2 commands on `/fmu/in/vehicle_attitude_setpoint` should be ignored if not in offboard mode.

Jaeyoung-Lim, 2023-12-09:
> Switching to Position mode while the offboard controller is still running (as in: still publishing its commands on the ros2 topic /fmu/in/vehicle_attitude_setpoint) does not work

This is the expected behavior of the microdds bridge. Therefore the reason for the check in px4-manipulation. You are directly injecting uorb messages into the message bus, so if you are sending conflicting information to the attitude controller, it will result in the behavior that you saw.

I agree that it might be useful to have a dedicated offboard topic to ignore setpoints, but this also has implocations on how transparent we want to keep offboard.

In any case, I would consider more of a feature request than a bug report.

ayhamalharbat, 2023-12-09:
> I am a bit confused that you mention that this is only true for VehicleAttitudeSetpoint, since this should be also for VehicleRateSetpoint, as well as other setpoints, depending on the offboard level.

I thought so at the time, I checked again now with `VehicleThrustSetpoint` + `VehicleTorqueSetpoint` , and `ActuatorMotors` commands, and I got the same problem. So you are right.
 
> I agree that it might be useful to have a dedicated offboard topic to ignore setpoints, but this also has implocations on how transparent we want to keep offboard.

I agree that the transparency of the bridge is paramount.

beniaminopozzan, 2023-12-12:
FYI @dagar , dedicated topics for offboard control

Jaeyoung-Lim, 2023-12-12:
@beniaminopozzan I have a draft of an offboard switch here: