PX4/PX4-Autopilot #22515: [Bug] FMU targets stopped building since new empy python package major update (3.3.4 -> 4.0)

DanielePettenuzzo, 2023-12-09:
### Describe the bug

I recently changed laptop and reinstalled my entire px4 toolchain on ubuntu 22.04 with the ubuntu.sh script. The installation script was successful but when try to build any fmu target it would fail due to a python error related to the empy package.

I noticed that on November 30, 2023 empy released a new major version so I tried downgrading from 4.0 to 3.3.4 using pip. This solved the issue so I am assuming that something changed in the empy api in this major update.

Unfortunately I don't have time to fix this so I wanted to raise this because I would assume that anyone installing or upgrading the toolchain in the next weeks will hit this.

### To Reproduce

1. Install toolchain with ubuntu.sh script after November 30, 2023
2. Build px4_fmu-v5x_default target (make px4_fmu-v5x_default)
3. See the following error:
```
Traceback (most recent call last):
  File "/home/daniele/src/px4/PX4_firmware_private/Tools/msg/px_generate_uorb_topic_files.py", line 261, in <module>
    generate_output_from_file(generate_idx, f, args.outputdir, args.package, args.templatedir, INCL_DEFAULT, all_topics)
  File "/home/daniele/src/px4/PX4_firmware_private/Tools/msg/px_generate_uorb_topic_files.py", line 171, in generate_output_from_file
    return generate_by_template(output_file, template_file, em_globals)
  File "/home/daniele/src/px4/PX4_firmware_private/Tools/msg/px_generate_uorb_topic_files.py", line 187, in generate_by_template
    em.RAW_OPT: True, em.BUFFERED_OPT: True})
AttributeError: module 'em' has no attribute 'RAW_OPT'
[20/1231] Generating ../../../platforms/nuttx/NuttX/nuttx/.config
ninja: build stopped: subcommand failed.
make: *** [Makefile:239: px4_fmu-v5x_default] Error 1
`

### Expected behavior

It should not fail and build successfully

### Flight Log

No log

### Software Version

Latest master from December 9, 2023

### Flight controller

PX4 FMU V5X

### Vehicle type

None

avcuenes, 2023-12-09:
This issue solved by this commit https://github.com/PX4/PX4-Autopilot/commit/3b041ef8d4f8493a166dcea0fb75ccca0e88b622

DanielePettenuzzo, 2023-12-09:
I missed that because my branch was slightly older than that. Thank you very much. I will close the issue.

closed 2023-12-09 by DanielePettenuzzo