PX4/PX4-Autopilot #22520: [CAN GPS] Dual F9P Heading Support for Non PX4 CAN Node GPS

vincentpoont2, 2023-12-12:
### Describe the bug

Using a Non PX4 CAN Node GPS, such as the Holybro DroneCAN F9P (Uses AP Periph). These Parameters below are not available in the **Component ###**:

- `CANNODE_SUB_MBD `
- `CANNODE_PUB_MBD`
- `GPS_UBX_MODE`
- `GPS_YAW_OFFSET`

Does this mean that Dual F9P GPS Heading is only supported if it is using PX4 CAN Node?

[image]

### To Reproduce

Connect a CAN GPS using AP Periph and go to QGC parameter.

### Expected behavior

See the parameter required to do CAN GPS Heading. 

### Flight Log

N/A

### Software Version

1.14 Stable

### Flight controller

Pixhawk 6X

### Vehicle type

None

ryanjAA, 2024-01-05:
@vincentpoont2 is there a can node firmware for new holybro mini f9p? Probably not too hard to make one. We could do flight testing of this hardware (any of the dual rtk but this ones makes most sense).

vincentpoont2, 2024-01-05:
> @vincentpoont2 is there a can node firmware for new holybro mini f9p? Probably not too hard to make one. We could do flight testing of this hardware (any of the dual rtk but this ones makes most sense). 

Hi Ryan, we have a few version of CAN F9P, which you referring to? One of them uses the NXP chipset and run this PX4 CAN FW https://github.com/PX4/PX4-Autopilot/tree/main/boards%2Fnxp%2Fucans32k146 

Two uses Ardupilot AP_periph fw. Those that run AP periph doesn't seem to have these parameter available.

vincentpoont2, 2024-01-09:
@julianoes @AlexKlimaj Any idea on what is required to allow F9P CAN GPS (running AP_Periph CAN FW) to use GPS Heading in PX4?

ryanjAA, 2024-01-09:
> Hi Ryan, we have a few version of CAN F9P, which you referring to? One of them uses the NXP chipset and run this PX4 CAN FW https://github.com/PX4/PX4-Autopilot/tree/main/boards%2Fnxp%2Fucans32k146 [...]

I was referring to the ultralight variant (see image) but it doesn't really matter, we can test any of them for functionality as long as basic functionality is there (which is the issue atm).

[image]

dirksavage88, 2024-01-10:
I’m wondering if it’s better long term to port these to PX4 CAN node. I’m not sure if AP periph is still using the bootloader that lets the app fall through (usually happens when user tries to flash an incorrect firmware binary) but if that’s the case and these boards have at least 512KB could be worth it. @ryanaa we did this with matek a while back

ryanjAA, 2024-01-10:
@dirksavage88 ya definitely agree. I think it would be a bit more work as some of these are nxp based but we can get in air testing done if we can get this operational.

julianoes, 2024-09-01:
So do I understand it correct that the actual issue is that PX4 doesn't expose/forward DroneCAN parameters? If so, it seems like that's a feature to implement/support in PX4 no matter what is running on a DroneCAN device.

AlexKlimaj, 2024-09-01:
You can edit the dronecan parameters through QGC.

julianoes, 2024-09-02:
@AlexKlimaj so it does work? Sorry I'm still confused.

dakejahl, 2024-09-02:
I think the issue is that PX4 hacks the heading 
https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/uavcannode/Publishers/GnssFix2.hpp#L135-L144
and ardupilot has a dedicated message
https://github.com/dronecan/DSDL/blob/993be80a62ec957c01fb41115b83663959a49f46/ardupilot/gnss/20002.Heading.uavcan
https://github.com/ArduPilot/ardupilot/blob/master/Tools/AP_Periph/gps.cpp#L230

julianoes, 2024-09-03:
I see, so we should probably support both then.

dakejahl, 2024-09-03:
Yeah, but I'm not sure if they're configured the same way as px4 with params 
https://ardupilot.org/copter/docs/common-gps-for-yaw.html

closed 2024-11-09 by dagar