PX4/PX4-Autopilot #22522: [Bug] Time jump detected. Resetting time synchronizer

frede791, 2023-12-12:
### Describe the bug

When running gazebo simulation with uxrce_dds_client enabled, the following error appears: `ERROR [timesync] Time jump detected. Resetting time synchroniser.`

### To Reproduce

On current main branch (commit: ece60b6165753c26f8cbc24cd6b2b2918ef8d8fe) run: 
`make px4_sitl gz_x500`
In a separate terminal start the uxrce_dds_client with: `MicroXRCEAgent udp4 -p 8888`
After a couple of seconds, the error will appear in the command line

### Expected behavior

No error should appear at all. While it doesn't break simulation, it is still unsightly. 

### Screenshot / Media

[image]

### Flight Log

NA

### Software Version

Current PX4 main branch.

### Flight controller

It's in simulation.

### Vehicle type

None

### Additional context

Reference previously by @PerFrivik and supposedly fixed in PR https://github.com/PX4/PX4-Autopilot/pull/22463. The belief is that the bug originated from [this PR](https://github.com/PX4/PX4-Autopilot/pull/22440).

PerFrivik, 2023-12-12:
I also still have the bug, tested on the newest main state.

beniaminopozzan, 2023-12-12:
I experienced this issue with gazebo simulation running close but not close enougth to real time factor 1.
PX4 takes the clock from Gazebo, XRCE-DDS agent uses the wall clock. Therefore agent and client runs with two different time paces and  the synchronizer resets when the accumulated drift gets too large.

You should stop the time synchronizer of XRCE-DDS and bind ROS2 to the Gazebo clock too
https://docs.px4.io/main/en/ros/ros2_comm.html#ros-gazebo-and-px4-time-synchronization

you can verify this worsening the gazebo real time factor, you should see the `ERROR [timesync] Time jump detected. Resetting time synchroniser` appear more frequently

closed 2024-01-16 by frede791