PX4/PX4-Autopilot #22536: [Bug] pixhawk 4 mini firmware not working

THOM-AwS, 2023-12-14:
### Describe the bug

I am trying to reflash my mini to get it working, but the firmware after flash does not allow you to calibrate the sensors, specifically the accelerometer and compass, I have tried a few times to get this running by flashing a build I made from source, and the auto downloaded version. 

Ive tried with apm2, and with qgc.

I get a red flashing light on the compass module light. sos tone, or ne-naw ne-naw ne-naw tone. 
[image]
[image]
[image]
[image]

### To Reproduce

1. pixhawk 4 mini, 
2. plug in, 
3. reflash firmware, 
4. upload completes, 
5. error messages as per screenshots.

I tried to rebuild in 1.9.0, and the build fails altogether. not sure which version would be appropriate, if not latest?

### Expected behavior

I expect the firmware to install and the fc to start working as normal.

### Screenshot / Media

attached above

### Flight Log

I dont get this far. 

### Software Version

latest build downloaded.

### Flight controller

pixhawk 4 mini

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

plugged into usb directly, 4 motor esc's, external gps. airframe is 250

sfuhrer, 2023-12-15:
Have you tried flashing v1.13 and 1.14? 1.9 is extremely old.

julianoes, 2023-12-16:
Have you tried resetting all params?