PX4/PX4-Autopilot #22542: mav_sys_id change causes unresponsive system

ryanjAA, 2023-12-14:
### Describe the bug

If you change mav_sys_id to ≠ 1 you can not reboot or change anything from QGC. 

manually rebooting works but then I needed to get out a debug probe to access the fmu to reset the param...

### To Reproduce

Used a few different versions of 1.14 release.

Setup a plane and then set mav_sys_id to 2 or 3.

Try and reboot via QGC and you should get a failed to reboot.

I would recommend doing a manual reboot since you need to set it from a debug board to fix it...

### Expected behavior

Should work as previously and have be able to have multiple units vehicles on the same network..

### Flight Log

none made

### Software Version

1.14 release

### Flight controller

v6x

### Vehicle type

Fixed Wing

### Additional context

Tried 3 different hardwares and software combos (and different versions of qgc between 4.2.9 and 4.3.0)

ryanjAA, 2023-12-14:
same on ark-v6x and main from a few days ago as per @AlexKlimaj

julianoes, 2023-12-16:
Did you flash from the release/1.14 branch or flash 1.14.0?

ryanjAA, 2023-12-16:
Release and @AlexKlimaj tried main. Same on both.

ryanjAA, 2024-01-11:
Pinging this for someone to take a look. Cant use more than one system on the same network/qgc instance...

junwoo091400, 2024-01-11:
> Should work as previously and have be able to have multiple units vehicles on the same network..

Do you know since when it was broken? I.e. what's the last version that multi vehicle connection works properly?

ryanjAA, 2024-01-11:
I unfortunately don’t know since the last production version we use is built on top of 1.11.3 which it still works. Skipped over 1.12 and 1.13.

ryanjAA, 2024-05-15:
@mrpollo not sure how to get this added to 1.15 board.

ryanjAA, 2024-08-16:
@mrpollo pinging you again on this since not sure who to ping as it's pretty crucial at the most basic levels of multiple aircraft on the same link.

I'd test this on 1.15 RCx but I am remote atm and don't have a debug probe with me if I soft-brick it.

@dagar

hamishwillee, 2024-08-21:
FYI, Ramon responded to this as "aware but not yet reproduced": https://discord.com/channels/1022170275984457759/1265375115722621010/1275675378970525696

mrpollo, 2024-08-22:
Hey @ryanjAA, I verified that this error exists within PX4. The issue is that we shouldn't allow Commander to propagate the changes until reboot. I have submitted a fix.

However, on QGC, my workaround is to close QGC and reopen it so it forgets your session because it will continue trying to ping the older sys_id.

closed 2024-08-23 by dagar

ryanjAA, 2024-08-23:
Tested the fix. Works as intended. 👍

PR #23601 (2024-08-23): Commander: lock down mav_sys_id and mav_comp_id
Only allow setting `MAV_SYS_ID` and `MAV_COMP_ID` when initializing commander, similar to the implementation in the mavlink module.

fixes #22542 

https://github.com/user-attachments/assets/147dd037-4317-4ac7-b9da-4ecb09c73171