PX4/PX4-Autopilot #22558: [Bug] uxrce_dds_client on main disconnects

AlexKlimaj, 2023-12-16:
### Describe the bug

It appears that there is a bug in main on the uxrce_dds_client where it will disconnect after only running for a few minutes. I'm not sure if this is related to the recent NUTTX DMA issues.

### To Reproduce

Run the uxrce_dds_client on main and it disconnects after a few minutes.

### Expected behavior

uxrce_dds_client stays connected.

### Flight Log

NA

### Software Version

main

### Flight controller

ARKV6X

### Vehicle type

None

beniaminopozzan, 2023-12-17:
Thanks @AlexKlimaj for spotting this. I cannot test it myself right now.
Could you determine if this issue is restricted to `px4_fmu-v6x` targets with serial communication or if it also affects other targets and upd communication?

AlexKlimaj, 2023-12-20:
It looks like I needed to revert these two commits and use @dagar DDS rework branch.

This PR
https://github.com/PX4/PX4-Autopilot/pull/22534

With these reverted. @davids5 
https://github.com/PX4/NuttX/commit/ed4814f6239097dde5eecf2b4fbd58661db84dda
https://github.com/PX4/NuttX/commit/3dc3cf522758d70ca2a07c156e5e02517650327d

davids5, 2023-12-20:
The driver should be looked at for how it uses the serial port. Before opening in non-blocking mode would block. Now that the driver is fixed it does not.

AlexKlimaj, 2023-12-20:
It does look like its opening it in non-blocking mode.

https://github.com/PX4/PX4-Autopilot/pull/22534/files#diff-c186a8a82bc9445fd93a8182d59a6d39855970f40612d9770855aea93df240eeR130

Mendeler, 2024-01-02:
It may be that the baud rate is too low. Try increasing it.

AlexKlimaj, 2024-01-02:
> It may be that the baud rate is too low. Try increasing it.

Not the issue. Running at 3Mbps.

closed 2024-01-23 by AlexKlimaj