PX4/PX4-Autopilot #22561: [Bug] px4-v1.14.0 VIO YAW SETPOINT seems flow the YAW ESTIMATE

HopeCollector, 2023-12-17:
### Describe the bug

[image]

the drone's yaw setpoint is following its estimation

here is the [log file](https://review.px4.io/plot_app?log=9dd559bc-78fc-4eb4-955c-a26f1c90b224)

[image]

in the picture above, we can find that the `Yaw Setpoint` is same as `Yaw Estimated`, and if we zoom in the plot

[image]

we can find that the setpoint value is try to following the estimated value?!!

### To Reproduce

1. setup the xrce_dds connection with the componion computer
2. change the following parameters
    * EKF2_GPS_CTRL: 0
    * EKF2_HGT_REF: 3(Vision)
    * EKF2_EV_CTRL: 15(fusion all data)
    * EKF2_EV_DELAY: 0
3. take off and hover
4. see the drone turn to left and right slowly (5 deg/s  i guss?)

### Expected behavior

- dorne should not noding
- `Yaw Setpoint` should be zero if just hover

### Flight Log

https://review.px4.io/plot_app?log=9dd559bc-78fc-4eb4-955c-a26f1c90b224

### Software Version

HW arch: PX4_FMU_V6X
HW type: V6X000003
HW version: 0x000
HW revision: 0x003
PX4 git-hash: b8c541dd7277ed735139d7d1bfb829d61fbe29fb
PX4 version: Release 1.14.0 (17694975)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: de41e7feaeffaec3ce65327e9569e8fdb553ca3d
Build datetime: Oct 19 2023 16:24:32
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 00060000000036333938313751120005001d
MCU: STM32H7[4|5]xxx, rev. Y

### Flight controller

skydroid T10

### Vehicle type

Multicopter

HopeCollector, 2023-12-17:
currently i'm working on these problem, any problem will be replied in a week

HopeCollector, 2023-12-18:
tested on v1.13.3 works fine, here is the [log file](https://review.px4.io/plot_app?log=e5037ed6-7002-4b77-8b96-e83b995ba258)

[image]

`Yaw Setpoint` is 0

i did not publish the tf between odom & odom_ned, base_link & base_link_ned, therefore the init yaw is 90

bresch, 2023-12-22:
Hi @HopeCollector . The reason why yaw tracking is inactive is because of this: https://github.com/PX4/PX4-Autopilot/blob/6be8cbe43966d57630e6743e163307c216444be7/src/modules/ekf2/EKF/ev_yaw_control.cpp#L166
In fact, yaw is not "aligned" (i.e.: wrt North) because the VIO frame isn't NED. This shouldn't be an issue for control but currently the logic disables the yaw control when yaw is not aligned. 

I'll work on a proper fix, but to unblock you, if you plan to use VIO only (no mag, no GNSS) you could either: change the flag I linked above and set it to `true` instead of `false` or send the VIO pose frame as NED.

bresch, 2023-12-22:
In fact, I checked in the code and the issue should already be fixed in `release/1.15-alpha1` , could you please verify that?

HopeCollector, 2023-12-23:
@bresch OK！I'll try 1.15 in next week and post the log file here as soon as possible, thanks for the great help

dirksavage88, 2023-12-28:
@bresch in EKF2 would it be okay to just disable VIO yaw and use mag as a heading source to get around this? I noticed this in 1.14 as well where yaw setpoints aren't being met, then I realized VIO is a local positioning source and without mag it has no reference heading.

bresch, 2024-01-03:
@dirksavage88 it should work, yes, but I can understand that one would prefer not to use a mag in some cases (e.g.: indoor operation)

HopeCollector, 2024-01-13:
> In fact, I checked in the code and the issue should already be fixed in `release/1.15-alpha1` , could you please test that?

@bresch  I'm sorry about the issue of testing the new version. After testing for a period of time, the code for this version still cannot run properly on my device, and this specific issue may occur
1. After updating from v1.13.3 to 1.15.0, flight control can be configured normally, but it cannot be connected after restarting
2. Reburning the firmware of v1.15.0 from v1.15.0, unable to burn properly, and the buzzer will not continuously prompt for successful burning

Because my problem doesn't seem to be a bug, I will close it.

Thank you very much for your help again. If there is anything I can help with, please feel free to contact me at any time

closed 2024-01-13 by HopeCollector

dagar, 2024-05-24:
> Strange that the bug report is closed as the bug is still present in all versions of 1.14.x and 1.15.x even with no VIO. It wasn't there in 1.13.x. Do I need to create a new bug report for this?

Can you share a log (ideally on main or 1.15x) and explain exactly what you're seeing? If you create a new issue with the details I can make it a release blocker for 1.15.0 at this point.

HopeCollector, 2024-05-25:
> Strange that the bug report is closed as the bug is still present in all versions of 1.14.x and 1.15.x even with no VIO. It wasn't there in 1.13.x. Do I need to create a new bug report for this?

> Hi @HopeCollector . The reason why yaw tracking is inactive is because of this: [...]

Hi @ArmadaAeronautics! 
According to bresch's explanation, I don't think this is a BUG. The problem is that the message received by the flight controller is not in the NED coordinate system.
Have you tried to send the positioning information to the flight controller according to the NED coordinate system? Does the plane still shake its head from side to side when hovering?