PX4/PX4-Autopilot #22563: Provide Gst Plugin and Camera Manager for Gazebo

frede791, 2023-12-18:
### Describe problem solved by the proposed feature

At the moment video from cameras cannot be streamed from Gazebo to PX4 and on to QGC. 

### Describe your preferred solution

Use either a gz-transport node or a ROS node to provide a GST plugin and a camera manager that can be integrated either into Gazebo or PX4.

### Describe possible alternatives

None

tnedev, 2024-02-26:
Are there any updates on this?
I'm interested in the GST plugin.

frede791, 2024-02-27:
No nobody is working on this right now as far as I know.

Meekohi, 2025-03-13:
What are other options for viewing the camera feed right now?

dakejahl, 2025-03-13:
a gz plugin has been added
https://github.com/PX4/PX4-Autopilot/pull/24475

camera manager will come later

closed 2025-03-13 by dakejahl