PX4/PX4-Autopilot #22564: Land Mode not Working as it Should after RC and GPS Loss[Bug]

Paspas03, 2023-12-18:
### Describe the bug

Hi everyone,
I am having an issue with PX4 going into Land Mode for Fixed Wing if GPS and RC Signal is Lost. Once this happens, the behaviour is not what is expected based on the applied parameters, rather the plane just continues on its current course, no control inputs are sent.

Does somebody know where the RC Loss Failsafe is handled? I expected to find it in the commander.cpp, but couldn't find it there.

### To Reproduce

1. Drone switched on
2. Turn GPS and RC off
3. See error

### Expected behavior

Following Prameters, should go into indefinite loiter. 

### Flight Log

No Log, as it is running on HITL Mode.

### Software Version

V1.13.2

### Flight controller

Cubepilot cubeorange

### Vehicle type

Fixed Wing

ryanjAA, 2023-12-20:
Try on 1.14 release since this is quite old now and changes unlikely to be made unless still present.

ryanjAA, 2023-12-20:
Also, a log would be needed. Is it actually going into failsafe, what safety settings are you using, etc

Paspas03, 2023-12-20:
Hi, 
thanks for your response. I unfortunately can not use the newest px4 version, but I have looked into it, to see if there were any code changes to fix the issue. Unfortunately I wasn't able to find the code that handles Manual control Loss Failsafe there either. 

Like I said, I unfortunately can not get a log as it isn't produced during HITL mode, but I can see it is definitely going into failsafe RTL mode when RC loss occurs and into land mode when both RC and GPS loss mode occurs. Unfortunately the land mode does not follow the parameter settings, but rather does absolutely nothing. 

I also managed to get it to Loiter if GPS is lost first and the RC by adding this code into the Run function of the comander.cpp, but for some reason it does not work if the failsafe order is switched:

//If GPS AND RC Connection is Lost, go into Loiter Mode

int sensor_sub_fd = orb_subscribe(ORB_ID(vehicle_status));
orb_set_interval(sensor_sub_fd, 200);
struct vehicle_status_s raw;
orb_copy(ORB_ID(vehicle_status), sensor_sub_fd, &raw);

if (raw.rc_signal_lost && !_status_flags.gps_position_valid) {
    PX4_DEBUG("GPS and Transmitter no longer valid. Activate Loiter Mode");
    px4_usleep(20000);
    main_state_transition(_status, commander_state_s::MAIN_STATE_AUTO_LOITER, _status_flags, _internal_state);
}

These are the safety parameters I have set:

param set-default NAV_RCL_ACT 2
param set-default COM_RC_LOSS_T 1 
param set-default COM_RCL_ACT_T 0 
param set-default PWM_MAIN_FAIL1 1500
param set-default PWM_MAIN_FAIL2 1500
param set-default PWM_MAIN_FAIL3 1500
param set-default PWM_MAIN_FAIL4 1500
param set-default PWM_MAIN_FAIL7 1000
param set-default PWM_MAIN_FAIL8 1440
param set-default CBRK_FLIGHTTERM 0
param set-default FD_FAIL_P 180
param set-default FD_FAIL_R 180
param set-default RTL_TYPE 0 
param set-default RTL_LAND_DELAY -1 
param set-default RTL_DESCEND_ALT 100 
param set-default RTL_RETURN_ALT 100 
param set-default FW_GPS_LT 5000
param set-default COM_POSCTL_NAVL 1

#From Failsafe State Machine Website
param set-default COM_ACT_FAIL_ACT 0
param set-default COM_FAIL_ACT_T 0
param set-default COM_IMB_PROP_ACT -1
param set-default COM_LKDOWN_TKO -1
param set-default COM_LOW_BAT_ACT 0
param set-default COM_OBL_RC_ACT 2
param set-default COM_QC_ACT -1
param set-default COM_RCL_EXCEPT 0
param set-default COM_SPOOLUP_TIME 0
param set-default NAV_DLL_ACT 0

I also looked into changing (neither of them changed anything)
COM_POSCTL_NAVL 0 and
FW_GPSF_LT 3600 to loiter for an hour once GPS is lost