PX4/PX4-Autopilot #22565: Vector asigment with vec.xy() = vec_xy() no longer works in 1.14 [Bug]

sverrevr, 2023-12-18:
### Describe the bug

In 1.14, the following assignment fails:
```c++
Vector3f lhs(1,2,3);
Vector3f rhs(4,5,6);
lhs.xy() = rhs.xy();
EXPECT_EQ(lhs, Vector3f(4,5,3));
```

```bash
[ RUN      ] MatrixVector3Test.Vector3
../../src/lib/matrix/test/MatrixVector3Test.cpp:70: Failure
Expected equality of these values:
  lhs
    Which is: 
               1               2               3

  Vector3f(4,5,3)
    Which is: 
               4               5               3

[  FAILED  ] MatrixVector3Test.Vector3 (0 ms)
```

So `lhs` is not getting assigned.

Note that the following test which is originally a part of `MatrixVector3Test.cpp` passes:
```c++
Vector2f g2(1, 3);
Vector3f g3(7, 11, 17);
g3.xy() = g2;
EXPECT_EQ(g3, Vector3f(1, 3, 17));
```

This bug is critical as this form of assignment is used in the code for example [here](https://github.com/PX4/PX4-Autopilot/blob/c63214bcaecdb84bdf07d7eb76cdef7a93498148/src/modules/flight_mode_manager/tasks/ManualPosition/FlightTaskManualPosition.cpp#L112-L121)
This line of code worked in 1.13 but no longer works in 1.14. 

I noticed this bug when testing 1.14 as the position lock no longer activated when using `MPC_POS_MODE = 0`.

### To Reproduce

Add the following lines of code to `PX4-Autopilot/src/lib/matrix/test/MatrixVector3Test.cpp`
```c++
Vector3f lhs(1,2,3);
Vector3f rhs(4,5,6);
lhs.xy() = rhs.xy();
EXPECT_EQ(lhs, Vector3f(4,5,3));
```
Run:
`make tests`

### Expected behavior

`lhs` should be assigned as it used to be in 1.13, the tests should pass.

### Flight Log

Not relevant as it can be tested with unit tests

### Software Version

Newest main when writing this issue: https://github.com/PX4/PX4-Autopilot/commit/c63214bcaecdb84bdf07d7eb76cdef7a93498148
And on 1.14 release

### Flight controller

N/A

### Vehicle type

None

sverrevr, 2023-12-18:
In 1.13 the position lock worked. This was broken by this commit which changed to using .xy instead of assigning each element separately. https://github.com/PX4/PX4-Autopilot/pull/20440/files#diff-6a93912921c62df6bced857c3dea68c3c275d25faa3f6c78866eafedc4e5612fR113

bresch, 2023-12-18:
Thanks for reporting, I'm looking into this issue.

MaEtUgR, 2023-12-19:
@sverrevr Thanks for spotting and reporting this in detail! We'll have a look.

sverrevr, 2023-12-20:
Good job! And thanks for solving this so rapidly  :) This solves the observed problem of the position lock not engaging. Position set points are now set when the joystick is released.

PR #22574 (2023-12-21): matrix: fix slice to slice assignment to do deep copy
### Solved Problem
@bresch warned me about this bug report yesterday: #22565
We started debugging and were puzzled why the assignment of `a.xy() = b.xy()` with two `Vector3f`s did not call `Slice::operator=()` which implements a deep copy of the submatrix in b, in this case the first two elements.

@jwidauer kindly spent some time analyzing the case. He found out by separating the steps and explicitly deleting constructors and the detailed output of clang that the autogenerated implicit default copy constructor of `Slice<Type, P, Q, M, N>`  [... cut]
closed 2023-12-21 by MaEtUgR