PX4/PX4-Autopilot #22569: [Bug] Checkout conflict after pulling newest commits with Tools/simulation/gz

notnitsuj, 2023-12-19:
### Describe the bug

After I git pulled the latest commits from origin/main to the main branch which seems to have added a new submodule, I couldn't check out to my development branch. The error message is below:

```
error: The following untracked working tree files would be overwritten by checkout:
	Tools/simulation/gz/models/advanced_plane/model.config
	Tools/simulation/gz/models/advanced_plane/model.sdf
	Tools/simulation/gz/models/omnicopter/model.config
	Tools/simulation/gz/models/omnicopter/model.sdf
	Tools/simulation/gz/models/px4vision/model.config
	Tools/simulation/gz/models/px4vision/model.sdf
	Tools/simulation/gz/models/rc_cessna/meshes/body.dae
	Tools/simulation/gz/models/rc_cessna/meshes/elevators.dae
	Tools/simulation/gz/models/rc_cessna/meshes/iris_prop_ccw.dae
	Tools/simulation/gz/models/rc_cessna/meshes/iris_prop_cw.dae
	Tools/simulation/gz/models/rc_cessna/meshes/left_aileron.dae
	Tools/simulation/gz/models/rc_cessna/meshes/left_flap.dae
	Tools/simulation/gz/models/rc_cessna/meshes/propeller_ccw.dae
	Tools/simulation/gz/models/rc_cessna/meshes/propeller_cw.dae
	Tools/simulation/gz/models/rc_cessna/meshes/right_aileron.dae
	Tools/simulation/gz/models/rc_cessna/meshes/right_flap.dae
	Tools/simulation/gz/models/rc_cessna/meshes/rudder.dae
	Tools/simulation/gz/models/rc_cessna/model.config
	Tools/simulation/gz/models/rc_cessna/model.sdf
	Tools/simulation/gz/models/standard_vtol/meshes/iris_prop_ccw.dae
	Tools/simulation/gz/models/standard_vtol/meshes/iris_prop_cw.dae
	Tools/simulation/gz/models/standard_vtol/meshes/x8_elevon_left.dae
	Tools/simulation/gz/models/standard_vtol/meshes/x8_elevon_right.dae
	Tools/simulation/gz/models/standard_vtol/meshes/x8_wing.dae
	Tools/simulation/gz/models/standard_vtol/model.config
	Tools/simulation/gz/models/standard_vtol/model.sdf
	Tools/simulation/gz/models/x500/model.config
	Tools/simulation/gz/models/x500/model.sdf
	Tools/simulation/gz/models/x500/thumbnails/1.png
	Tools/simulation/gz/models/x500/thumbnails/2.png
	Tools/simulation/gz/models/x500/thumbnails/3.png
	Tools/simulation/gz/models/x500/thumbnails/4.png
	Tools/simulation/gz/models/x500/thumbnails/5.png
	Tools/simulation/gz/models/x500_depth/model.config
	Tools/simulation/gz/models/x500_depth/model.sdf
	Tools/simulation/gz/models/x500_depth/thumbnails/1.png
	Tools/simulation/gz/models/x500_depth/thumbnails/2.png
	Tools/simulation/gz/models/x500_depth/thumbnails/3.png
	Tools/simulation/gz/models/x500_depth/thumbnails/4.png
	Tools/simulation/gz/models/x500_depth/thumbnails/5.png
	Tools/simulation/gz/models/x500_vision/model.config
	Tools/simulation/gz/models/x500_vision/model.sdf
	Tools/simulation/gz/tools/avl_automation/README.md
	Tools/simulation/gz/tools/avl_automation/avl_out_parse.py
	Tools/simulation/gz/tools/avl_automation/avl_steps.txt
	Tools/simulation/gz/tools/avl_automation/input.yml
	Tools/simulation/gz/tools/avl_automation/input_avl.py
	Tools/simulation/gz/tools/avl_automation/process.sh
	Tools/simulation/gz/tools/avl_automation/templates/advanced_lift_drag_template.sdf
	Tools/simulation/gz/tools/avl_automation/templates/control_surface.sdf
	Tools/simulation/gz/worlds/default.sdf
	Tools/simulation/gz/worlds/windy.sdf
Please move or remove them before you switch branches.
Aborting
```

When I used `git status` it only showed my uncommitted changes in the other branch and none of the above.

### To Reproduce

I don't know which commit I last pulled but it should be `f38fe24a98e05e5ecada347c32e108a12afb03aa` as I pull everyday.

```
git checkout main
git pull
git submodule update --recursive
git checkout dev
```

### Expected behavior

Checkout normally like everyday

### Flight Log

N/A

### Software Version

main 1.14

### Flight controller

N/A

### Vehicle type

None

PerFrivik, 2023-12-20:
Has to do with #22467, @frede791 is on it.

frede791, 2023-12-20:
@notnitsuj Hey so the issue is that gz is now a submodule on main whereas on your own branch `dev` you are still using the old folder structure meaning Git will complain when you are trying to switch branches, as there are files in your working directory that are not tracked by Git, but they exist in the branch you're trying to switch to. Depending on what you have changed, you can force the checkout using `-f` or make a new branch from main and cherry-pick your own commits.

notnitsuj, 2023-12-20:
Yeah I was able to bypass all that as my changes are unrelated to `gz`. Just wanted to let you guys know in case there are any hidden conflicts with making `gz` a submodule. Thanks for your replies.

closed 2023-12-20 by notnitsuj