PX4/PX4-Autopilot #22573: 1.14.0 release Optical flow position with ros 0,0,0 bug and 1.15.0 beta velocity estimate error [Bug]

lgh5054, 2023-12-20:
### Describe the bug

1.14.0 release When using only optical flow, /mavros/local_position/odom is continuously initialized to 0 when viewed in rviz.
1.15.0 beta When using only optical flow Velocity estimation error continues to occur

### To Reproduce

1.14.0 realese 
Enable optical flow
enable tf in ros rviz and subscribe to /mavros/local_position/odom

1.15.0 later
Enable optical flow

### Expected behavior

1.14.0
Continuously initialized from rviz to takeoff starting point

1.15.0
Velocity estimation error continues to occur message

### Flight Log

not

### Software Version

1.14.0 release and 1.15.0 later

### Flight controller

kakute h7 mini

### Vehicle type

Multicopter

bresch, 2023-12-22:
Which optical flow module are you using?
Please provide a log. Without that it would be really difficult to investigate.

lgh5054, 2023-12-27:
@bresch kakute h7 does not support log save.
However, it is clear that there is a problem with the current 1.14.0 version

I am using clover coex optical flow.

It operates optical flow using an rpi cam.
https://clover.coex.tech/en/optical_flow.html

bresch, 2024-01-03:
> it is clear that there is a problem with the current 1.14.0 version

Did it work on 1.13 then?

> kakute h7 does not support log save.

Ok, maybe I could investigate a bit if you can provide the output of:

`listener sensor_optical_flow`
`listener vehicle_optical_flow`
`listener estimator_optical_flow_vel`
`listener estimator_status_flags`

lgh5054, 2024-01-08:
@bresch 
this is beta 1.15.0 version

TOPIC: sensor_optical_flow
 sensor_optical_flow
    timestamp: 774012575 (0.011503 seconds ago)
    timestamp_sample: 774012574 (1 us before timestamp)
    device_id: 11665678 (Type: 0xB2, MAVLINK:1 (0x01))
    pixel_flow: [-0.00006, -0.00012]
    delta_angle: [0.00001, -0.00001, -0.00001]
    distance_m: 0.00000
    integration_timespan_us: 35904
    error_count: 0
    max_flow_rate: nan
    min_ground_distance: nan
    max_ground_distance: nan
    delta_angle_available: True
    distance_available: False
    quality: 12
    mode: 0

nsh> listener vehicle_optical_flow

TOPIC: vehicle_optical_flow
 vehicle_optical_flow
    timestamp: 796105794 (0.022564 seconds ago)
    timestamp_sample: 796105784 (10 us before timestamp)
    device_id: 11665678 (Type: 0xB2, MAVLINK:1 (0x01))
    pixel_flow: [-0.00032, -0.00005]
    delta_angle: [-0.00001, 0.00001, -0.00001]
    distance_m: nan
    integration_timespan_us: 32031
    max_flow_rate: 8.00000
    min_ground_distance: 0.08000
    max_ground_distance: 100.00000
    quality: 18

nsh> listener estimator_optical_flow_vel

TOPIC: estimator_optical_flow_vel
 estimator_optical_flow_vel
    timestamp: 812509661 (0.033617 seconds ago)
    timestamp_sample: 812445791 (63870 us before timestamp)
    vel_body: [-0.00106, -0.00025]
    vel_ne: [-0.00107, -0.00019]
    flow_rate_uncompensated: [-0.00113, 0.00464]
    flow_rate_compensated: [-0.00128, 0.00532]
    gyro_rate: [0.00014, -0.00068, -0.00016]
    gyro_bias: [-0.00599, -0.00661, 0.00053]
    ref_gyro: [0.00397, 0.00555, -0.00120]

nsh> listener estimator_status_flags

TOPIC: estimator_status_flags
 estimator_status_flags
    timestamp: 825258319 (0.651650 seconds ago)
    timestamp_sample: 825196947 (61372 us before timestamp)
    control_status_changes: 80
    fault_status_changes: 0
    innovation_fault_status_changes: 0
    cs_tilt_align: True
    cs_yaw_align: False
    cs_gps: False
    cs_opt_flow: True
    cs_mag_hdg: False
    cs_mag_3d: False
    cs_mag_dec: False
    cs_in_air: False
    cs_wind: False
    cs_baro_hgt: False
    cs_rng_hgt: True
    cs_gps_hgt: False
    cs_ev_pos: False
    cs_ev_yaw: False
    cs_ev_hgt: False
    cs_fuse_beta: False
    cs_mag_field_disturbed: False
    cs_fixed_wing: False
    cs_mag_fault: False
    cs_fuse_aspd: False
    cs_gnd_effect: False
    cs_rng_stuck: False
    cs_gps_yaw: False
    cs_mag_aligned_in_flight: False
    cs_ev_vel: False
    cs_synthetic_mag_z: False
    cs_vehicle_at_rest: True
    cs_gps_yaw_fault: False
    cs_rng_fault: False
    cs_inertial_dead_reckoning: False
    cs_wind_dead_reckoning: False
    cs_rng_kin_consistent: True
    cs_fake_pos: False
    cs_fake_hgt: False
    cs_gravity_vector: False
    cs_mag: False
    cs_ev_yaw_fault: False
    cs_mag_heading_consistent: False
    cs_aux_gpos: False
    fs_bad_mag_x: False
    fs_bad_mag_y: False
    fs_bad_mag_z: False
    fs_bad_hdg: False
    fs_bad_mag_decl: False
    fs_bad_airspeed: False
    fs_bad_sideslip: False
    fs_bad_optflow_x: False
    fs_bad_optflow_y: False
    fs_bad_vel_n: False
    fs_bad_vel_e: False
    fs_bad_vel_d: False
    fs_bad_pos_n: False
    fs_bad_pos_e: False
    fs_bad_pos_d: False
    fs_bad_acc_bias: False
    fs_bad_acc_vertical: False
    fs_bad_acc_clipping: False
    reject_hor_vel: False
    reject_ver_vel: False
    reject_hor_pos: False
    reject_ver_pos: False
    reject_yaw: False
    reject_airspeed: False
    reject_sideslip: False
    reject_hagl: False
    reject_optflow_x: False
    reject_optflow_y: False

nsh> param show EKF2_OF*
Symbols: x = used, + = saved, * = unsaved
x + EKF2_OF_CTRL [318,640] : 1
x   EKF2_OF_DELAY [319,641] : 0.0000
x + EKF2_OF_GATE [320,642] : 30.0000
x   EKF2_OF_N_MAX [321,643] : 0.3000
x   EKF2_OF_N_MIN [322,644] : 0.1000
x   EKF2_OF_POS_X [323,645] : 0.0000
x   EKF2_OF_POS_Y [324,646] : 0.0000
x   EKF2_OF_POS_Z [325,647] : 0.0000
x + EKF2_OF_QMIN [326,648] : 10
x   EKF2_OF_QMIN_GND [327,649] : 0

lgh5054, 2024-01-08:
1.15.0 beta works fine.

There is a problem with version 1.14.0 stable.

bresch, 2024-01-08:
Thanks 

> There is a problem with version 1.14.0 stable.

Could you then please send the same outputs using 1.14.0 stable?

lgh5054, 2024-01-09:
@bresch this is v1.14.0 stable 
 
TOPIC: sensor_optical_flow
 sensor_optical_flow
    timestamp: 644473117 (0.003292 seconds ago)
    timestamp_sample: 644473116 (1 us before timestamp)
    device_id: 11665678 (Type: 0xB2, MAVLINK:1 (0x01))
    pixel_flow: [0.00012, 0.00011]
    delta_angle: [-0.00003, -0.00003, 0.00001]
    distance_m: 0.00000
    integration_timespan_us: 32008
    error_count: 0
    max_flow_rate: nan
    min_ground_distance: nan
    max_ground_distance: nan
    delta_angle_available: True
    distance_available: False
    quality: 4
    mode: 0

nsh> listener vehicle_optical_flow

TOPIC: vehicle_optical_flow
 vehicle_optical_flow
    timestamp: 660501195 (0.032866 seconds ago)
    timestamp_sample: 660501184 (11 us before timestamp)
    device_id: 11665678 (Type: 0xB2, MAVLINK:1 (0x01))
    pixel_flow: [0.00052, 0.00026]
    delta_angle: [-0.00003, 0.00000, 0.00001]
    distance_m: nan
    integration_timespan_us: 36006
    max_flow_rate: 8.00000
    min_ground_distance: 0.20000
    max_ground_distance: 10.00000
    quality: 0

nsh> listener estimator_optical_flow_vel

TOPIC: estimator_optical_flow_vel
 estimator_optical_flow_vel
    timestamp: 674102263 (0.022738 seconds ago)
    timestamp_sample: 674084676 (17587 us before timestamp)
    vel_body: [0.00020, 0.00050]
    vel_ne: [-0.00050, 0.00020]
    flow_uncompensated_integral: [0.00006, -0.00001]
    flow_compensated_integral: [0.00008, -0.00003]
    gyro_rate: [-0.00057, 0.00078, 0.00060]
    gyro_rate_integral: [-0.00001, 0.00002, 0.00001]
nsh> listener estimator_status_flags

TOPIC: estimator_status_flags
 estimator_status_flags
    timestamp: 692012388 (0.485598 seconds ago)
    timestamp_sample: 692000829 (11559 us before timestamp)
    control_status_changes: 11
    fault_status_changes: 0
    innovation_fault_status_changes: 0
    cs_tilt_align: True
    cs_yaw_align: True
    cs_gps: False
    cs_opt_flow: False
    cs_mag_hdg: False
    cs_mag_3d: False
    cs_mag_dec: False
    cs_in_air: False
    cs_wind: False
    cs_baro_hgt: False
    cs_rng_hgt: True
    cs_gps_hgt: False
    cs_ev_pos: False
    cs_ev_yaw: False
    cs_ev_hgt: False
    cs_fuse_beta: False
    cs_mag_field_disturbed: False
    cs_fixed_wing: False
    cs_mag_fault: False
    cs_fuse_aspd: False
    cs_gnd_effect: False
    cs_rng_stuck: False
    cs_gps_yaw: False
    cs_mag_aligned_in_flight: False
    cs_ev_vel: False
    cs_synthetic_mag_z: False
    cs_vehicle_at_rest: True
    cs_gps_yaw_fault: False
    cs_rng_fault: False
    cs_inertial_dead_reckoning: True
    cs_wind_dead_reckoning: False
    cs_rng_kin_consistent: True
    cs_fake_pos: True
    cs_fake_hgt: False
    cs_gravity_vector: False
    fs_bad_mag_x: False
    fs_bad_mag_y: False
    fs_bad_mag_z: False
    fs_bad_hdg: False
    fs_bad_mag_decl: False
    fs_bad_airspeed: False
    fs_bad_sideslip: False
    fs_bad_optflow_x: False
    fs_bad_optflow_y: False
    fs_bad_vel_n: False
    fs_bad_vel_e: False
    fs_bad_vel_d: False
    fs_bad_pos_n: False
    fs_bad_pos_e: False
    fs_bad_pos_d: False
    fs_bad_acc_bias: False
    fs_bad_acc_vertical: False
    fs_bad_acc_clipping: False
    reject_hor_vel: False
    reject_ver_vel: False
    reject_hor_pos: False
    reject_ver_pos: False
    reject_yaw: False
    reject_airspeed: False
    reject_sideslip: False
    reject_hagl: False
    reject_optflow_x: False
    reject_optflow_y: False

nsh> param show EKF2_OF*
Symbols: x = used, + = saved, * = unsaved
x   EKF2_OF_CTRL [337,609] : 0
x + EKF2_OF_DELAY [338,610] : 0.0000
x   EKF2_OF_GATE [339,611] : 3.0000
x + EKF2_OF_N_MAX [340,612] : 0.2000
x + EKF2_OF_N_MIN [341,613] : 0.0500
x   EKF2_OF_POS_X [342,614] : 0.0000
x   EKF2_OF_POS_Y [343,615] : 0.0000
x   EKF2_OF_POS_Z [344,616] : 0.0000
x + EKF2_OF_QMIN [345,617] : 10
x   EKF2_OF_QMIN_GND [346,618] : 0

bresch, 2024-01-16:
`EKF2_OF_CTRL [337,609] : 0` so `cs_opt_flow: False`. There is no position nor velocity aiding, this is why the position is constantly resetting.

julianoes, 2024-09-09:
No more follow up, presumably this is resolved.

closed 2024-09-09 by julianoes