PX4/PX4-Autopilot #22582: [Bug] moment calculated with wrong formula in Actuator Effectiveness Rotors

duy1102002, 2023-12-22:
### Describe the bug

in methods: ActuatorEffectivenessRotors::computeEffectivenessMatrix
moment is calculated as 
`matrix::Vector3f moment = ct * position.cross(axis) - ct * km * axis;`

but in v1.13.3 px_generate_mixers
torgue is calculated as
`
def torque_matrix(center, axis, dirs, Ct, Cm):
    '''
    Compute torque generated by rotors
    '''
    # normalize rotor axis
    ax = axis / np.linalg.norm(axis, axis=1)[:, np.newaxis]
    torque = Ct * np.cross(center, ax) - Cm * ax * dirs
    return torque
`

So, the Ct should used be
`matrix::Vector3f moment = ct * position.cross(axis) - km * axis * directions;
 // directions was not used
`

### To Reproduce

Not availiable

### Expected behavior

Not availiable

### Flight Log

Not availiable

### Software Version

V1.14.0 master

### Flight controller

px4-fmu_v3

### Vehicle type

None

bresch, 2023-12-22:
Hi, the directions is now included in the sign of `km`. See https://github.com/PX4/PX4-Autopilot/blob/6be8cbe43966d57630e6743e163307c216444be7/ROMFS/px4fmu_common/init.d/airframes/4019_x500_v2#L32C28-L32C28 for example.

duy1102002, 2023-12-25:
> Hi, the directions is now included in the sign of `km`. See https://github.com/PX4/PX4-Autopilot/blob/6be8cbe43966d57630e6743e163307c216444be7/ROMFS/px4fmu_common/init.d/airframes/4019_x500_v2#L32C28-L32C28 for example.

Thanks for the reply. But another question is `
matrix::Vector3f moment = ct * position.cross(axis) - ct * km * axis;
`
what is the "- ct * km" stands for?  Is ct nessary to multiply ?