PX4/PX4-Autopilot #22587: [Bug] Two PX4 boards cannot be used simultaneously in offboard mode

serkanMzlm, 2023-12-25:
### Describe the bug

"While operating with a single PX4, it can be easily used in offboard mode with ROS2 in both simulation and real-world scenarios. When attempting to create two vehicles, following the specified steps in the simulation environment allows both vehicles to be used. However, in the real world, I encounter an issue; I cannot use two PX4s simultaneously because both boards broadcast on the same topic names

### To Reproduce

Power up the drone
sudo MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 921600

### Expected behavior

I expect each PX4 vehicle to broadcast on different topic names, similar to how it is done in the simulation

### Flight Log

https://logs.px4.io/plot_app?log=6d25c321-00a0-43d4-9260-0cd254e3dc91

### Software Version

1.14

### Flight controller

Holybro Pixhawk 4 (Discontinued)

### Vehicle type

None

beniaminopozzan, 2023-12-27:
Hi @serkanMzlm , please see the answer in the [discord server](https://discord.com/channels/1022170275984457759/1189151466427002950/1189510513072283678).

When starting the client from parameter selection [UXRCE_DDS_CFG](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#UXRCE_DDS_CFG), the namespace cannot be changed (because there are no string parameters in PX4). Therefore all topics have namespace hardcoded namespace `/fmu/[in,out]/`. However, you can modify your namespace adding another prefix by starting the client without parameters.

serkanMzlm, 2023-12-28:
@beniaminopozzan You're welcome for your responses

closed 2023-12-28 by serkanMzlm