PX4/PX4-Autopilot #22590: Request: QSPI flash storage

David-OConnor, 2023-12-30:
### Describe problem solved by the proposed feature

PX4 does not support QSPI flash storage

### Describe your preferred solution

Support QSPI flash storage

### Describe possible alternatives

Continuing to only support basic SPI and SDIO

### Additional context

I am requesting support for QuadSPI (QSPI) flash support on ArduPilot for use with Flash storage, as an alternative to SPI flash and SD Cards. QSPI is a protocol explicitly designed for use with flash memory.

Reasons:
- QSPI is a faster alternative to SPI flash, and can use memory mapping
- QSPI has no downsides compared to SPI
- All STM32 MCUs supported by AP have a QSPI peripheral
- Most (all?) memory chips supported by AP are capable of QSPI.

As an engineer, I would, choose QSPI over SPI for flash storage for any project that uses an MCU with QSPI support. So, it was surprising that AP supports traditional SPI for flash, but not QSPI.

[STM32H743 Reference manual](https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) See Chapter 23: QUAD-SPI Interface

Igor-Misic, 2024-01-23:
This board has QSPI-enabled memory so you can use it as a reference for your board. 
https://github.com/PX4/PX4-Autopilot/tree/main/boards/spracing/h7extreme/src
It loads firmware from Flash into RAM in memory-mapped mode and switches to normal mode to write parameters on it.

David-OConnor, 2024-01-23:
That's a bit different, no? That is about putting program memory and parameters on flash, vice logging.

Igor-Misic, 2024-01-23:
I didn't know you were looking for logging. 
I believe this should all be configurable through Nuttx config, enabling QSPI for storage devices and ideally combining it with littlefs. The only change in PX4 is to change the path from `/fs/microsd` to `/fs/qspi `

NXP guys have already done it on their NXP MCUs.

w407022008, 2024-01-25:
[Holybro Kakute H7 mini](https://docs.px4.io/main/en/flight_controller/kakuteh7mini.html). this board has used QSPI onboad flash ([W25N01GV](https://datasheet.lcsc.com/lcsc/1809200036_Winbond-Elec-W25N01GVZEIG_C88868.pdf)), which use the same mcu as official board e.g. pixhawk 6.