PX4/PX4-Autopilot #22599: [Bug] STM32 Serial Mavlink on Main Disconnects

AlexKlimaj, 2024-01-03:
### Describe the bug

On current main with the recent DMA changes, I cannot keep serial mavlink connected to a companion computer. It will run for a few minutes before locking up and needing a reboot of the flight controller. With the DMA changes reverted, it stays connected indefinitely. 

### To Reproduce

Run mavlink serial on main with an STM32H7 connected directly to a companion computer.

### Expected behavior

Stays connected. 

### Flight Log

NA

### Software Version

main after nuttx st dma changes

### Flight controller

ARKV6X

### Vehicle type

None

dagar, 2024-01-03:
@davids5 @niklaut @AlexKlimaj can we hop on a call to discuss this as soon as possible? If mavlink is broken on STM32H7 in main this is pretty serious.

davids5, 2024-01-04:
> @davids5 @niklaut @AlexKlimaj can we hop on a call to discuss this as soon as possible? If mavlink is broken on STM32H7 in main this is pretty serious.

@dagar Sure - let's coordinate on `discord maintainers`  I have openings 8-12 EST today

AlexKlimaj, 2024-01-04:
I think we need to revert these commits and re-evaluate them.

https://github.com/PX4/NuttX/commit/ed4814f6239097dde5eecf2b4fbd58661db84dda
https://github.com/PX4/NuttX/commit/3dc3cf522758d70ca2a07c156e5e02517650327d

davids5, 2024-01-04:
No @AlexKlimaj.  A non blocking operations should not block. Period! Test it. Pull CTS false the system will hang!  Wrong is wrong! Let's Fix the root cause.

niklaut, 2024-01-08:
I'm confused that TELEM1,2,3 are all configured with flow control in NuttX, but Mavlink status only shows the flow control for TELEM3. Flow control is also off when it's working? Is the flow control even physically used by the other communication side?

AlexKlimaj, 2024-01-08:
I tested with and without flow control connected. The lock up occurs in both cases.

davids5, 2024-01-08:
@AlexKlimaj Did you terminate the CTS Lpw?

AlexKlimaj, 2024-01-08:
No it was floating. But the mavlink driver is disabling flow control. Shouldn't the state of the pin not matter in that instance?

davids5, 2024-01-08:
Should be is the Key word here.

Please get a scope on the Serial lines and HS and let's see what your CPU is seeing

niklaut, 2024-01-08:
Also, [looking at the code, hardware flow control can automatically be disabled in `FLOW_CONTROL_AUTO` mode?](https://github.com/PX4/PX4-Autopilot/blob/20129e63facab129ad31fa693d1397e7458afaa4/src/modules/mavlink/mavlink_main.cpp#L730). The flow control mode is controlled by the `-z` (on), `-Z` (off)`, or none (auto) flags of `mavlink start`. So it may start with flow control on, but then gets switched off at some time…

AlexKlimaj, 2024-01-08:
Yes but this lock up occurs on mavlink and dds. dds always has flow control turned off. Serial TX will run for a short period of time then just stop outputting.

davids5, 2024-01-08:
Off and on are time relative. At boot up it is on until turned off. If there is trapped data when it was on, that could cause the issues. Please try to pull the line Low from boot and see if it affects it.

AlexKlimaj, 2024-01-09:
https://github.com/PX4/PX4-Autopilot/pull/22614/files

With serial TX DMA disabled, the DDS client is still working 9 hours later.

davids5, 2024-01-09:
@AlexKlimaj @niklaut has been trying to reproduce the failure you are seeing. But can not repro it yet. Do you have a Salea Logic analyzer? If So I can add Probes to the Serial driver that you can run in the failing system. The LA is connected to the PWM OUT pins. This way we can see the state and see if there is a race we are not seeing.

niklaut, 2024-01-10:
I've been running the serial_test for several hours and cannot reproduce this on our Skynode X config (FMUv6x with the same config as yours). 

```
/dev/ttymxc3: count for this session: rx=579016685, tx=571762328, rx err=6024
/dev/ttymxc3: TIOCGICOUNT: ret=0, rx=579107680, tx=571859587, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
```

I'm also working on understanding max throughput better with the serial test via tracing. However, DMA utilization seems very high on first glance (we use a 3Mbps UART link to the companion). I'll investigate further. [orbetto.perf.gz -> ui.perfetto.io](https://github.com/PX4/PX4-Autopilot/files/13891416/orbetto.perf.gz)

davids5, 2024-01-10:
@niklaut What about just using mavlink?

PetervdPerk-NXP, 2024-01-15:
@niklaut you've tested asymmetrical flow? For example only tx is transmitting and rx doesn't send anything. I've got a hunch the issues come from the fact that mavlink and xrce don't constantly flood both tx rx like the serial test does.

niklaut, 2024-01-16:
I've tried a number of things including running serial test asymmetrically. I just cannot reproduce this and I haven't gotten any reports of this failing in the field in our test vehicles. I'll revisit this once I can get a handle on it.

AlexKlimaj, 2024-01-16:
Just got a binary built on GCC 9.3.1 to fail. Testing the same firmware built with GCC 10.2.1 now. Have a scope hooked up to TX/RX and I'm waiting to see if it fails. When it does fail, TX is idling high and QGC is sending the heartbeat.

Igor-Misic, 2024-01-18:
Here is one of the way to reproduce this issue relatively quickly.
- start mavlink on one of the UARTs with enabled **TX DMA** (_example: mavlink start -d /dev/ttyS2 -b 921600 -r 290000 -m onboard_low_bandwidth -x_) 
- connect QGC to FMU over previously set UART
- start downloading one log file with 200 MB or more and wait

In 90% of time I wasn't able to finish download.

davids5, 2024-01-18:
> Here is one of the way to reproduce this issue relatively quickly.
> 
> * start mavlink on one of the UARTs with enabled **TX DMA** (_example: mavlink start -d /dev/ttyS2 -b 921600 -r 290000 -m onboard_low_bandwidth -x_)
> * connect QGC to FMU over previously set UART
> * start downloading one log file with 200 MB or more and wait
> 
> In 90% of time I wasn't able to finish download. In my case, I don't have flow control enabled nor does it exist on hardware.

@Igor-Misic What HW are you testing on and what is connected?

Igor-Misic, 2024-01-18:
> @Igor-Misic What HW are you testing on and what is connected?

Unfortunately, it is a development board with STM32H7 without USB so I was forced to use UART to connect QGC. 
Unresponsive/blocked UART led me to this thread. 
Maybe my issue is not the same as this one but it seems I can't reproduce it if I don't use Tx DMA.  Hm...

davids5, 2024-01-18:
I have added PROBEs instrumentation on the V6X. N.B the PWM is disabled and used for PROBES.

https://github.com/PX4/PX4-Autopilot/tree/main-dnm-probes-h7-serial

The Second commit - may be the solution. Can you test this branch?

Igor-Misic, 2024-01-18:
> The Second commit - may be the solution. Can you test this branch?

So far it works!

davids5, 2024-01-19:
@AlexKlimaj @Igor-Misic - Thank you both for testing.

After sleeping on it, I realized there was still a window that could be interrupted. It is much smaller, then it used to be, but still exited. 

I have closed the window. Please test main using [this NuttX](https://github.com/PX4/NuttX/tree/px4_firmware_nuttx-10.3.0%2B-pr-h7-serial-dma-busy)

davids5, 2024-01-19:
Here is what was happening:

Normal

[image]

Caused the hang

[image]

davids5, 2024-01-22:
@Igor-Misic & @AlexKlimaj did you test 56aa08568f8eb84f8096114e1a9485fb0d3b0d18 ?

Igor-Misic, 2024-01-22:
I have applied your patch and since then haven't seen any issues. Also, it works ~20% faster than previously.

niklaut, 2024-01-22:
I'm soak testing this change on our CI for the next 12h, so we'll know if it impacts our setup too.

AlexKlimaj, 2024-01-22:
Looking good!

niklaut, 2024-01-23:
There's no change in behavior in our CI. Which goes in line with my inability to reproduce this bug on our hardware setup. So… merge?

MaEtUgR, 2024-01-23:
Fix https://github.com/PX4/PX4-Autopilot/pull/22667 was merged, please reopen if the problem persists.