PX4/PX4-Autopilot #22610: [Bug] PIXHAWK6X-RT: "Preflight Fail: Crash dumps present on SD" is logged

muramura, 2024-01-06:
### Describe the bug

PIXHAWK6X-RT was shipped at the end of December 2023.
Multiple hard faults occur in a short period of time.
Occurred during a health check.
It has also occurred during a flight at SIH.

Even SIH does not work properly with this PIXHAWK6X-RT.

### To Reproduce

1. Build and upload the PX4-Autopilot source hash "cf840ff3731d8bebf65e79bd8c9c7bbd8d29d404" to PIXHAWK6X-RT.

2. After uploading, turn the power back on.

3. Energize the power for at least 30 minutes.

### Expected behavior

1. repeat the reboot.

2. the orange LED stays on.

3. cannot communicate with QGC via USB connection.

### Screenshot / Media

[image]

[image]

### Flight Log

[hardfault_log] -- 2024-01-06-12:29:45 Begin Fault Log --
System fault Occurred on: 2024-01-06-12:29:45
 Type:Hard Fault in file:irq/irq_unexpectedisr.c at line: 59 running task: wq:SPI2
 FW git-hash: cf840ff3731d8bebf65e79bd8c9c7bbd8d29d404
 Build datetime: Jan  6 2024 20:59:08
 Build url: localhost 
 Processor registers: from 0x2026d76c
 r0:0x20267df0 r1:0x20267d4c  r2:0x0002b72e  r3:0x39090005  r4:0x20267df0  r5:0x2026d864 r6:0x00000000 r7:0x00000080
 r8:0x2026d874 r9:0x00000001 r10:0x2026d8ac r11:0x00000000 r12:0x2026d508  sp:0x2026d840 lr:0x0000474d pc:0x39090004
 xpsr:0x21000000 basepri:0x000000f0 control:0x00000004
 exe return:0xffffffe9
 Fault status registers: from NVIC
 cfsr:0x00000100 hfsr:0x00000000  dfsr:0x00000000  mmfsr:0x00000000  bfsr:0x00000000 afsr:0x00000000 abfsr:0x00000000 
 IRQ stack: 
  top:    0x20249368
  sp:     0x20249320 Valid
  bottom: 0x20248b68
  size:   0x00000800
  used:   0x000000d8
 User stack: 
  top:    0x2026d8d8
  sp:     0x2026d840 Valid
  bottom: 0x2026cf98
  size:   0x00000940
  used:   0x00000720
Interrupt sp memory region, stack pointer lies within stack
0x20249ac8 0x0620dd00
0x20249ac4 0x00000030
0x20249ac0 0x06212900
0x20249abc 0x00000030
0x20249ab8 0x06217500
0x20249ab4 0x00000030
0x20249ab0 0x0621c100
0x20249aac 0x00000030
0x20249aa8 0x0631c500
0x20249aa4 0x00000030
0x20249aa0 0x0631c500
[... 1954 lines cut ...]
0x2026d0b8 0xefdeadbe
0x2026d0b4 0xefdeadbe
0x2026d0b0 0xefdeadbe
0x2026d0ac 0xefdeadbe
0x2026d0a8 0xefdeadbe
0x2026d0a4 0xefdeadbe
0x2026d0a0 0xefdeadbe
0x2026d09c 0x00deadbe
0x2026d098 0x00000000
0x2026d094 0x00000000
0x2026d090 0xadbe0000
0x2026d08c 0xadbeefde
[hardfault_log] -- 2024-01-06-12:29:45 END Fault Log --

### Software Version

cf840ff3731d8bebf65e79bd8c9c7bbd8d29d404

### Flight controller

PIXHAWK6X-RT

### Vehicle type

Multicopter

davids5, 2024-01-08:
@muramura 
Please post the log from boot. 
What devices are connected. 
What HW base board are you using?

muramura, 2024-01-08:
@davids5 san.
I have built the source on HOLYBRO PIXHAWK6X-RT and only energized it.

I am hoping to build a master PX4-Autopilot and hope no hard faults occur.

davids5, 2024-01-08:
I have a PR that will come in today, that you should use to test with https://github.com/PX4/PX4-Autopilot/pull/22608

muramura, 2024-01-08:
I am verifying the operation with SIH.
During automatic flight, a hard fault occurs.
I have investigated and it is occurring with pointer variables.
I am dealing with them one by one.
Hard fault occurs during auto flight.

[image]

[image]

muramura, 2024-01-24:
@davids5 san. 
I have two PIXHAWK6X-RT.
I was instructed by HOLYBRO to install and check the shipping firmware.
I installed the shipping firmware on all PIXHAWK6X-RTs.
Result,
1. a HARD FAULT occurred on PIXHAWK6X-RT where a HARD FAULT had occurred.
2. PIXHAWK6X-RT, which did not have a HARD FAULT, did not have a HARD FAULT.

It is uploaded below.
https://github.com/PX4/PX4-Autopilot/issues/22660

PetervdPerk-NXP, 2024-01-25:
> I have two PIXHAWK6X-RT.
I was instructed by HOLYBRO to install and check the shipping firmware.
I installed the shipping firmware on all PIXHAWK6X-RTs.
Result,
> 1.    a HARD FAULT occurred on PIXHAWK6X-RT where a HARD FAULT had occurred.
> 2.    PIXHAWK6X-RT, which did not have a HARD FAULT, did not have a HARD FAULT.

@muramura  I'm not sure what's meant here, are you saying that you have 2 PIXHAWK 6X-RT flight controllers and that only 1 of these suffer from the faults? Or do both flight controllers suffer from these faults you're reporting?

muramura, 2024-01-25:
Yes, it is.
I have two PIXHAWK6X-RTs.
One of them is a HARD FAULT.

muramura, 2024-02-10:
I purchased a third PIXHAWK6X-RT.
I let it heat run on its own for over an hour.
This PIXHAWK6X-RT did not cause a HARD FAULT.

I was convinced.
The PIXHAWK6X-RT that generates a HARD FAULT is a hard defect.

davids5, 2024-02-10:
@vincentpoont2 FYI