PX4/PX4-Autopilot #22625: [Bug] HITL simulation motors not spinning PX4 v1.14.0

vacoff, 2024-01-11:
### Describe the bug

The first problem with hitl was observed in the Airsim simulator when upgraded to the 1.14 version of PX4. Until that with 1.13.3 version Airsim worked just fine ( had a problem with mission uploading, but unrelated to this issue). Then we thought the problem was in the Airsim side, so tried to run the Gazebo hitl simulation, but problem was also present there. As everything connects to each-other without problem, commands also reach and even we can see signal to the motor commands in the logs. But the motors is not spinning. I suspect that the cause might be the dynamic mixer allocation related somehow.

### To Reproduce

1. Upload 1.14.0 version to the flight controller from QGC
2. `git clone --branch release/1.14 https://github.com/PX4/PX4-Autopilot.git --recursive`
3. Follow the [HITL Simulation guide](https://docs.px4.io/v1.14/en/simulation/hitl.html).

### Expected behavior

The uav, quadcopter in this case to spin its motors and fly according to the rc input or mission.

### Screenshot / Media

[image]
[image]

### Flight Log

From gazebo: https://logs.px4.io/plot_app?log=0fc1709f-9ab5-4cbc-bb72-fb596af114f8

From Airsim: https://logs.px4.io/plot_app?log=11c84cd4-e76a-4c20-81fe-5afaaeee8ce7

### Software Version

v1.14.0

### Flight controller

Cube Orange with ADSB carieer board

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Cube Orange plugged into the pc using Serial and other connections are default UDP ports.

thuhangkhuat, 2024-01-19:
Hello @vacoff, I also had the same problem. Have you solved it?

vacoff, 2024-01-19:
Hi, unfortunately no, still waiting response from maintainers.

jnomikos, 2024-02-01:
Also, when running the VTOL HITL simulation specifically, I am spammed with MAG and COMPASS timeouts. Might be an unrelated issue, I am not sure.

Also, when running SIH on the cube orange plus, I get a preflight fail of every sensor missing, and I don't see the drone on the map.

As another note, for HITL and SIH, actuators are never getting assigned by default. Maybe this is normal? I am new to HITL.

NanyongGit, 2024-03-20:
> Also, when running the VTOL HITL simulation specifically, I am spammed with MAG and COMPASS timeouts. Might be an unrelated issue, I am not sure.
>
> Also, when running SIH on the cube orange plus, I [...]

I face the exact same problems with all the sensors missing and missing drone. Have you found a workaround?

VladSmirnov777, 2024-03-20:
I'm using cubeorangeplus.And next thing solved this issue:
In file boards/cubepilot/cubeorangeplus/default.px4board add line:
CONFIG_MODULES_SIMULATION_PWM_OUT_SIM=y
Now I'm running HIL Standard VTOL with Gazebo Classic and everything runs as expected (according to https://docs.px4.io/v1.14/en/simulation/hitl.html).

NanyongGit, 2024-03-20:
> I'm using cubeorangeplus.And next thing solved this issue: In file boards/cubepilot/cubeorangeplus/default.px4board add line: CONFIG_MODULES_SIMULATION_PWM_OUT_SIM=y Now I'm running HIL Standard [...]

Hi VladSmirnov, thanks so much for replying. Can you share the exact steps you took because I'm still unable to replicate it. I'm using the cube black so I added the `CONFIG_MODULES_SIMULATION_PWM_OUT_SIM=y `into my boards/px4/fmu-v3/default.px4board then I did `make px4_fmu-v3_default ` to compile it, and then `make px4_fmu-v3_default upload `to upload it to my pixhawk. 

I then did a make clean again and followed the steps in the documentation. After running gazebo or jmavsim, the motors are still not turning. Any idea why?

Thank you once again for your reply.

VladSmirnov777, 2024-03-20:
MAVLink Console nsh>help ===> Check if pwm_out_sim module exists.
Check QGroundControl MAVLink Inspector window HIL_ACTUATOR_CONTROLS message existing
And first of all I have checked in Wireshark if any udp messages sent to the port 14550 (listen port of Gazebo Classic).
And one important thing:I'm using Gazebo Classic,not new Gazebo.

NanyongGit, 2024-03-21:
Hi Vlad, thanks for replying. i just checked, pwm_out_sim does not exist here, there's pwm_input, and pwm_out but not the sim one. 

In the Mavlink Inspector, HIL_ACTUATOR_CONTROLS exists and there's a frequency. I'm using Gazebo classic as well.

VladSmirnov777, 2024-03-21:
May I ask if your motors will move if you move the sliders under Actuators in QGC?
Good question.I have never checked it.
No,it's not moving under Actuators but after created some flight and start mission it's moving.

NanyongGit, 2024-03-21:
Thank you! @VladSmirnov777 Will try it again

NanyongGit, 2024-03-21:
Sorryy @VladSmirnov777, I restarted again using my Pixhawk cube black, and the HIL_ACTUATOR_CONTROLS don't show up when I'm running jmavsim and QGC. It shows up when it's connected directly though, do you happen to know why and how to fix this?

FilBk, 2024-04-11:
Hi all, @jnomikos any update about the spamming  message of MAG and COMPASS timeouts in HITL VTOL simulation? 
I do have the same situation, with  just same setup as described by @vacoff .

Beyond CONFIG_MODULES_SIMULATION_PWM_OUT_SIM=y which made the plane actually  spin props, but i had no proper flight.
In also addeed  in /boards/cubepilot/cubeorange/default.px4board file before building anything the following: 
CONFIG_MODULES_SIMULATION_SENSOR_BARO_SIM =y
CONFIG_MODULES_SIMULATION_SENSOR_GPS_SIM =y
CONFIG_MODULES_SIMULATION_SENSOR_MAG_SIM =y

For some reason, something probably linked with the sensor errors which keeps spamming, is obstructing the flightcontroller to perform  a simple assigned mission, which is working just fine in SITL.

BehnoushHatami, 2024-06-14:
Hello everyone.
I am experiencing an issue with the HITL simulation and seeking some advice.
I am using Pixhawk 6C as my flight controller with PX4 1.14.3. The gazebo version is 11.10.2, QGroundControl is v4.3.0, and I selected the airframe as Holybro X500 V2 (since it is mentioned that it is compatible with HITL since it is a generic quadcopter airframe).
I successfully set up my environment according to the HITL documentation. The iris model is visible and correctly loaded in Gazebo, and it appears that the flight controller is connected to both Gazebo and QGroundControl without any issues. However, I am unable to arm the vehicle using either my transmitter or directly through QGroundControl…