PX4/PX4-Autopilot #22633: The error occurs when using FlightGear for software-in-the-loop simulation in v1.14 and later.

747640013, 2024-01-12:
### Describe the bug

I use FlightGear for fixed-wing software-in-the-loop simulation. However, after running the command "make px4_sitl_nolockstep flightgear_rascal-electric," I encountered the following error:

ERROR [simulator_mavlink] Number of simulated accelerometers 80 out of range. Max: 3.

This issue started occurring after version 1.14, but it was not present in version 1.13.3. I am looking to resolve this problem, but I am having difficulty identifying the issue. Does anyone have any suggestions?

### To Reproduce

1.cd /the path/PX4-Autopilot
2.make px4_sitl_nolockstep flightgear_rascal-electric
3.waiting for FG load
4. The error is as follows:
ERROR [simulator_mavlink] Number of simulated accelerometers 80 out of range. Max: 3.
or ERROR [simulator_mavlink] Number of simulated accelerometer 96 out of range. Max: 3

### Expected behavior

provide some suggestions

### Flight Log

https://review.px4.io/plot_app?log=089bc4a9-4f92-40e9-9775-5051d74f5f44

### Software Version

v1.14

### Flight controller

no

### Vehicle type

None

747640013, 2024-01-13:
I seem to have found the cause of the issue. In PX4-Autopilot/Tools/simulation/flightgear/flightgear_bridge/vehicle_state.cpp, the rcehicleState::getSensorMsg function lacks initialization for sensor_msg.id.

closed 2024-01-13 by 747640013

bmklein5, 2024-07-24:
@747640013 how did you initialize it?