PX4/PX4-Autopilot #22634: [Bug] Offboard control not working for 1.14 version

jodusan, 2024-01-12:
### Describe the bug

I am trying to run offboard control example from Jetson Orin NX board on a real quadcopter with Pixhawk 6X running px4 1.14  connected via ethernet link https://github.com/PX4/px4_ros_com/tree/main/src/examples/offboard_py with gps as location but it is not working. Micro-dds is working fine I see and list all topics but it just doesnt allow me to arm the drone.

It switches to offboard mode but it cannot arm and reports:
[image]

At first I was trying to do it with realsense camera similar to t265 which provides tracking and would send odometry messages to:
/fmu/in/vehicle_visual_odometry

Offboard control signal to:
/fmu/in/offboard_control_mode

```
 def offboard_timer_callback(self):
        """Publish the offboard control mode."""
        msg = OffboardControlMode()
        msg.position = True
        msg.velocity = False
        msg.acceleration = False
        msg.attitude = False
        msg.body_rate = False
        msg.timestamp = int(Clock().now().nanoseconds / 1000)
        self.offboard_control_mode_pub.publish(msg)
```

And trajectory setpoint to:
/fmu/in/trajectory_setpoint

```
    def publish_position_setpoint(self, x: float, y: float, z: float):
        """Publish the trajectory setpoint."""
        msg = TrajectorySetpoint()
        msg.position = [x, y, z]
        msg.velocity = [0.1, 0.1, 0.1]
        msg.yaw = 1.57079  # (90 degree)
        msg.timestamp = int(Clock().now().nanoseconds / 1000)
        self.trajectory_setpoint_pub.publish(msg)
        self.logger.info(f"Publishing position setpoints {[x, y, z]}")
```

But whatever I tried I never managed to arm it and it would always fail with the error as in the image above. 

Similar problem is mentioned here:
https://github.com/PX4/PX4-Autopilot/issues/22592

### To Reproduce

I cannot run even the simplest offboard example, try running any on the internet:

ie:

https://github.com/Jaeyoung-Lim/px4-offboard

https://github.com/PX4/px4_ros_com/blob/main/src/examples/offboard/offboard_control.cpp

### Expected behavior

I expect the drone to arm and take off reaching height -10m and then land

### Flight Log

No flight log available

### Software Version

PX4 1.14
Ubuntu 20.04 Galactic
QGroundControl 4.3.0

### Flight controller

Pixhawk 6X, holybro

### Vehicle type

Multicopter

antonio-sc66, 2024-01-13:
I use PX4 v1.14 with ROS1 and position and velocity setpoints are working as before, so it looks like it's either a setup related issue or a ROS2<->PX4 related issue

jodusan, 2024-01-13:
@antonio-sc66 do you use mavros or xrce agent (i dont think it xan even work with ros1)? It could be due to ros2 somehow. What flags did you have to set?  Did it work on first try by following official docs or was some more googling necessary? Thanks for replying!

antonio-sc66, 2024-01-15:
@jodusan I tend to use ROS1 Noetic with MAVROS. Latest combo I've tested was PX4 v1.14.0 and MAVROS [v1.17.0](https://github.com/mavlink/mavros/releases/tag/1.17.0). 

Altough it's an older setup and **ROS2 is considered the preferred way**, I still find ROS1 interaction with PX4 to be a good option in terms of interface stability.

If you want to give it a try you can use the following links:
- [PX4 - ROS1 guide](https://docs.px4.io/main/en/ros/ros1.html)
- [PX4 - MAVROS Offboard Python](https://docs.px4.io/main/en/ros/mavros_offboard_python.html)

For ROS2, read the following guide:
- [PX4ROS 2 User Guide](https://docs.px4.io/main/en/ros/ros2_comm.html)

FastRTPS was used in v1.13, but it has been replaced in v1.14 in favour of [uXRCE-DDS](https://docs.px4.io/main/en/middleware/uxrce_dds.html) middleware. There's a migration guide available in that link. Maybe the example is not up to date with the latest changes. You might be able to find more help via Discord or the PX4 forum.

closed 2024-01-22 by jodusan