PX4/PX4-Autopilot #22641: [Bug] A bug occurs during landing when using only the lidar sensor for altitude. Manual and automatic are the same.

lgh5054, 2024-01-16:
### Describe the bug

A bug occurs during landing when using only the lidar sensor for altitude. Manual and automatic are the same.

### To Reproduce

ekf2_ev_ctrl 0
ekf2_gps_ctrl 0
ekf2_baro_ctrl 0
ekf2_rng_ctrl 2
ekf2_mag_type none

### Expected behavior

If you raise the throttle in altitude mode and then slowly lower the throttle after flight, you will lose altitude.
The same thing happens during auto landing

### Screenshot / Media

[image]
failsafe activate fallback to stabilized while losing altitude

### Flight Log

https://logs.px4.io/plot_app?log=59031a84-cb53-41b0-bbc2-0bdaf27a90aa

### Software Version

developer version but same 1.14.0 stable

### Flight controller

pixhawk 6c and kakuteh7 mini

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

tf mini lidar to serial and i2c I tested them all and they all have the same symptoms.

bresch, 2024-01-16:
Hi, you might want to set the minimum range to a smaller value (e.g.: `EKF2_MIN_RNG` = 0.03)
[image]

lgh5054, 2024-01-16:
@bresch Hello When land speed is set to 0.2, it hops to an altitude of 1m or more after landin

loghatVector, 2026-03-09:
It seems like you're experiencing issues with altitude control during landing when using the lidar sensor. A good next step would be to analyze the flight log you provided for any anomalies in the lidar readings during the descent phase. Pay particular attention to the altitude data and the corresponding throttle commands to see if there are any discrepancies or unexpected behaviors. 

Additionally, you might want to check the configuration of the lidar sensor in the parameters to ensure it's correctly set up for altitude measurement. If you haven't already, consider using LogHat (loghat.app) for a more detailed analysis of the .bin file, which can help identify any sensor fusion issues or control anomalies.