PX4/PX4-Autopilot #22649: [Bug] Rover firmware problem - no controls in mission

leszekambro, 2024-01-17:
### Describe the bug

Hello, I'm trying to implement autopilot into a full scale farm tractor. I want to use only one autopilot output to control the tractor steering wheel (without throttle). I prepared the appropriate motor with gearbox and driver. I am able control the motor manually from the qgc or or with using RC transmitter. I set up the mission with standard waypoints at altitude 0, the mission is activated in qgc but there is no reaction on the autopilot outputs. The firmware I am using is 1.14. I also tried to load older versions eg. 1.12, 1.13 but with no changes. I tested it on different hardware versions - fmuv2, orange cube pilot etc. and stil no reaction on the outpus. Maybe there is someone who could give me some advice and suggest where to look for the problem?

### To Reproduce

In Rover firmaware upload mission and after execution there is no reaction on the outputs.  

### Expected behavior

Control signal on the outputs steering actuator/motor

### Flight Log

https://review.px4.io/plot_app?log=6be5f582-c4bf-44fc-898f-b75538102a1b
https://review.px4.io/plot_app?log=84122da9-c2a6-47bb-99c2-b4e8745353fb

### Software Version

1.14

### Flight controller

CUBEPILOT_CUBEORANGE

### Vehicle type

Rover

sfuhrer, 2024-01-23:
@PerFrivik pinging you as it's about rover, maybe you can help here

PerFrivik, 2024-01-23:
Taking a look.

MaEtUgR, 2024-01-24:
@leszekambro So the problem in both logs seems to be that the rover control module `rover_pos_control` is not running because it's not built by default neither for fmu-v2 nor cube orange 😬 That can be seen from any performance counter like `rover_pos_control: cycle:` being missing and no related parameters like e.g. `GND_SP_CTRL_MODE` being logged.

The default build configurations are:
fmu-v2: https://github.com/PX4/PX4-Autopilot/blob/release/1.14/boards/px4/fmu-v2/default.px4board
cube orange: https://github.com/PX4/PX4-Autopilot/blob/release/1.14/boards/cubepilot/cubeorange/default.px4board
and neither of them contain `CONFIG_MODULES_ROVER_POS_CONTROL=y` most likely because of flash constraint reasons.

The easiest way for you to have it running would be to use the fmu-v2 rover build which of course includes that controller:
https://github.com/PX4/PX4-Autopilot/blob/release/1.14/boards/px4/fmu-v2/rover.px4board#L10
I don't know if that's currently deployed automatically or needs to be built, we're discussing how to deploy such specialiced builds because we run against flash limits on many boards for various reasons and separate builds would be useful but only if easily available.

Does that help? Do you need detailed steps to get that version? Would you prefer to use the cube orange?

leszekambro, 2024-01-24:
Hi, thank you for your reply. I uploaded the firmware automatically from
QGC so it looks like it doesn't deploy properly in automatic mode. Are you
able to provide for me the builds for nevest stable px4 release (1.14) for
fmu-v2 and cube orange (I mean the .px4 files to upload it via qgc)? This
way we could check for fast whether your diagnosis is correct.

WBR,
Leszek

IdroG-24, 2024-04-09:
@MaEtUgR I have the same problem of @leszekambro and I use a Pixhawck 6X. I read that the problem is related to the lack of 
CONFIG_MODULES_ROVER_POS_CONTROL=y in the .px4board file. Can you explain to me how to proceed to modify the file?
Thank you

chfriedrich98, 2025-02-11:
Closing this issue since enabling the rover modules is now properly documented in the [user guide](https://docs.px4.io/main/en/frames_rover/#flashing-the-rover-build).

closed 2025-02-11 by chfriedrich98