PX4/PX4-Autopilot #22654: [Bug] python module empy-4.0.1 causing AttributeError: module 'em' has no attribute 'RAW_OPT'

NumbNutN, 2024-01-18:
### Describe the bug

The python module **empy** version v4.0.1 seems to cause an error when I built the PX4 firware for **Holybro Kakuteh7 Mini**

### To Reproduce

`$ make holybro_kakuteh7_default`
```
[0/1013] git submodule src/modules/mavlink/mavlink
[2/1013] Generating uORB topic headers
FAILED: uORB/topics/uORBTopics.hpp uORB/topics/action_request.h uORB/topics/actuator_armed.h
...
Traceback (most recent call last):
  File "/mnt/d/Coding/slam/PX4-Autopilot/msg/tools/px_generate_uorb_topic_files.py", line 544, in <module>
    generate_output_from_file(
  File "/mnt/d/Coding/slam/PX4-Autopilot/msg/tools/px_generate_uorb_topic_files.py", line 184, in generate_output_from_file
    return generate_by_template(output_file, template_file, em_globals)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/d/Coding/slam/PX4-Autopilot/msg/tools/px_generate_uorb_topic_files.py", line 354, in generate_by_template
    em.RAW_OPT: True, em.BUFFERED_OPT: True})
    ^^^^^^^^^^
AttributeError: module 'em' has no attribute 'RAW_OPT'
```

### Expected behavior

The version requirement for module **empy** may need to be further refined.

https://github.com/PX4/PX4-Autopilot/blob/b60e73c76f8da7e2d18569f71163b08ca8e98fac/Tools/setup/requirements.txt#L5

### Flight Log

None

### Software Version

v1.13.0

### Flight controller

holybro kakuteh7

### Vehicle type

Multicopter

julianoes, 2024-01-18:
I'm confused. The version is already fixed in requirements.txt. However, it looks like you have the wrong version installed, right?

NumbNutN, 2024-01-20:
My fault. It happens in tag v1.13.0. I refer to wrong code.

closed 2024-01-20 by NumbNutN