PX4/PX4-Autopilot #22659: [Bug] px4_sitl gz_standard_vtol spawns with camera under the world

Ryanf55, 2024-01-19:
### Describe the bug

To run `gz_standard_vtol` for terrain-navigation, it's spawning the camera under the terrain. You have to click to go to the vehicle instead of it spawning the camera at the right location.

### To Reproduce

```bash
cd PX4-AutoPilot
./Tools/setup/ubuntu.sh
# Source environment variables mentioned in PX4-Autopilot/Tools/simulation/gz/worlds/davosdorf.sdf
export PX4_HOME_LAT=46.8132056
export PX4_HOME_LON=9.8479538
export PX4_HOME_ALT=1562.0
export PX4_GZ_VERBOSE=4
export PX4_GZ_MODEL_POSE="0,0,1562,0,0,0"
export PX4_GZ_WORLD=davosdorf
# TODO remove these variables and make it use a proper aligned world.

make px4_sitl gz_standard_vtol
```

[image]

### Expected behavior

The camera spawns in where it ends up if you click "standard_vtol_0" -> "Move To"

Then, it looks like this - correct.

[image]

### Flight Log

No need for a log because it's not flying yet.

### Software Version

 https://github.com/srmainwaring/PX4-AutoPilot/tree/prs/pr-hinwil-testing-rebased  hash 1b8a937ab268419ad245340b01676c827a71e236
 
 It's slightly off main.

### Flight controller

sitl

### Vehicle type

Hybrid VTOL

### How are the different components wired up (including port information)

SITL only

### Additional context

@Jaeyoung-Lim has ideas on how to fix it.

frede791, 2024-01-25:
@Ryanf55 You will probably have to write a script to get the correct elevation for your mesh. You can then maybe move the model down by that in the sdf file (something like `<pose> 0 0 -elevation 0 0 0 </pose>` and that would fix your camera pose issue. Alternatively, adjust the starting position by the elevation and the camera pose accordingly. 
In the past I have used blender and its bpy APi to find the elevation and then written the elevation into an sdf "template"

frede791, 2024-02-15:
@Ryanf55 any update on this?

Ryanf55, 2024-02-15:
Hello,

I haven't gotten a chance to write a script for it. I would have thought that mesh knows its location and the UAV can be expressed relative to the mesh. 

If/when I can give this a try, I'll report back. However, if you are interested to try to solve it, I'm happy to test what you had in mind out!

frede791, 2024-02-15:
@Ryanf55 I think your issue can be fixed by adjust the `camera_pose` tag as seen below. In your sdf file you should have the GzScene3D plugin and within that the `camera_pose` tag. Set that to a value higher than the spawn altitude of your vtol and it should sit correctly.

```shell
<plugin name='3D View' filename='GzScene3D'>
        <gz-gui>
          <title>3D View</title>
          <property type='bool' key='showTitleBar'>0</property>
          <property type='string' key='state'>docked</property>
        </gz-gui>
        <engine>ogre2</engine>
        <scene>scene</scene>
        <ambient_light>0.5984631152222222 0.5984631152222222 0.5984631152222222</ambient_light>
        <background_color>0.8984631152222222 0.8984631152222222 0.8984631152222222</background_color>
        <camera_pose>-6 0 1570 0 0.5 0</camera_pose>
      </plugin>
```

Here I adjusted the camera to sit 8 metres above your defined VTOL height.

frede791, 2024-02-15:
The reason I mentioned the script is because if you are auto-generating your sdf files then it would obviously be nice to set the camera pose correctly without manually changing it. I have done this by using an sdf template with placeholder values which I replace when generating my sdf files with an elevation variable. The value for this variable I get from Google's Elevation API.

Ryanf55, 2024-02-15:
> ```shell
>  1570
> ```

Do you know of a way to make the <camera_pose> relative to the VTOL rather than the world origin?

frede791, 2024-02-15:
Ok so I investigated a bit but I am not sure it is possible: with regular `pose` tags you can tether links to other links by specifying `relative_to`. So something like this

```shell
<pose relative_to="link1"> 0 0 1 0 0 0 </pose>
```
is valid for say link2. However, this doesn't seem to exist in the source code for the <camera_pose> tag. Hence i don't think it is possible.

Ryanf55, 2024-02-15:
Great info. I'm going to create a request upstream to add support for `relative_to` in the `camera_pose` and link it here. In the mean time, I can just hard code the offset. 

Note - I didn't create this world file; I'm not sure if was scripted before, but I'm sure we can follow in the footsteps of other projects generating SDF with jinja. 

You can use `gdallocationinfo` any dataset by the way to get the elevation of a specific point of a dataset.

closed 2024-08-21 by dagar