PX4/PX4-Autopilot #22660: [Bug] PIXHAWK6X-RT generates HARD FAULT

muramura, 2024-01-21:
### Describe the bug

I purchased two PIXHAWK6X-RT from HOLYBRO.
I connected the QGC to check the operation.
One of them generates a HARD FAULT in untimed.
The other one generates a HARD FAULT.

I believe it is an initial failure of the FC.

### To Reproduce

1. launch QGC
2. Connect PIXHAWK6X-RT to your PC.
3. Connect for more than 30 minutes.

### Expected behavior

Restart when a HARD FAULT occurs.

### Flight Log

HARD FAULT
https://review.px4.io/plot_app?log=25dad990-4315-43bf-8021-e79b4d47065c

NOT HARD FAULT
https://review.px4.io/plot_app?log=8f3832c5-2f10-47c0-aa08-3aebe1411b61

### Software Version

V1.15.0alpha

### Flight controller

PIXHAWK6X-RT

### Vehicle type

Multicopter

vincentpoont2, 2024-01-23:
@davids5 @PetervdPerk-NXP @dk7xe Any idea what can be causing this?

muramura, 2024-01-24:
I have two PIXHAWK6X-RT.
I was instructed by HOLYBRO to install and check the shipping firmware.
I installed the shipping firmware on all PIXHAWK6X-RTs.
Result,
1. a HARD FAULT occurred on PIXHAWK6X-RT where a HARD FAULT had occurred.
2. PIXHAWK6X-RT, which did not have a HARD FAULT, did not have a HARD FAULT.

It is uploaded below.
https://github.com/PX4/PX4-Autopilot/issues/22660

davids5, 2024-01-25:
@muramura Please test current main.

muramura, 2024-01-25:
I uploaded the latest MAIN to PIXHAWK6X-RT where HARD FAULT occurs.
A HARD FAULT has occurred.

Can this FC be determined to be a hardware failure?

https://review.px4.io/plot_app?log=d248f3e0-1d98-4b57-a354-6fa14c0876f6

davids5, 2024-01-25:
@muramura What is connected to the Setup? A picture would help.

muramura, 2024-01-25:
I am connecting PIXHAWK6X-RT alone to my PC.
PIXHAWK6X-RT is only connected to a PC.

PIXHAWK6X-RT<-(USB)->PC(UBUNTU, QGC)

PetervdPerk-NXP, 2024-01-26:
@muramura  I think it could be related to SD card corruption. Could you swap the sd cards from both boards and see if the crash moves over to the other FMU?

muramura, 2024-01-26:
I put the MICRO SD card of the PIXHAWK6X-RT, which does not generate a HARD FAULT, into the PIXHAWK6X-RT, which does.

The result was that a HARD FAULT occurred.

Log:
https://review.px4.io/plot_app?log=feccc2c9-ff17-4416-827c-d106e7104bad

PetervdPerk-NXP, 2024-01-26:
Okay I think I've might found a procedure to get your PIXHAWK6X-RT into a correct state.

1. Remove the SD card
2. Startup PIXHAWK6X-RT
3. Open up a nsh console and type `mtd erase`
4. Format the SD card
5. Re-insert the SD card

muramura, 2024-01-26:
I have implemented the procedure.

The result was a HARD FAULT occurred.

https://review.px4.io/plot_app?log=95af7a19-74c2-4bfe-93bb-cea90b22391b

vincentpoont2, 2024-01-30:
> @PetervdPerk-NXP san.
> 
> I put the MICRO SD card of the PIXHAWK6X-RT, which does not generate a HARD FAULT, into the PIXHAWK6X-RT, which does.
> 
> The result was that a HARD FAULT occurred.
> 
> Log: https://review.px4.io/plot_app?log=feccc2c9-ff17-4416-827c-d106e7104bad

@muramura , you are saying you have two Pixhawk 6x-RT, and only one of them have hardfault. The other one is working without issue, is that correct?

muramura, 2024-01-30:
Dear HOLYBRO STAFF

Yes, that is correct.
One does not experience a single HARD FAULT.

dk7xe, 2024-02-06:
I guess this is the same https://github.com/PX4/PX4-Autopilot/issues/22610 ?

davids5, 2024-02-06:
@muramura - Please test current main.

muramura, 2024-02-10:
I purchased a third PIXHAWK6X-RT.
I let it heat run on its own for over an hour.
This PIXHAWK6X-RT did not cause a HARD FAULT.

I was convinced.
The PIXHAWK6X-RT that generates a HARD FAULT is a hard defect.

https://review.px4.io/plot_app?log=865c086b-5ea4-4b12-97c2-931e5f04545e