PX4/PX4-Autopilot #22663: [Bug] FollowMe feature does not seem to respect SYS_VEHICLE_RESP

Fraisa, 2024-01-22:
### Describe the bug

I've been testing Follow Me for quite some time now, and now I wanted to make it less aggressive as I'm using a mid-size drone. When tuning SYS_VEHICLE_RESP, FLW_TGT_RS and the limits to acceleration and jerk I noted no difference to the drone behavior. 

An uneducated guess: The code should have a "mid-step" where the control system "We need this acceleration/jerk/etc" is limited to whatever max-acceleration/jerk/etc is setup. I feel like that could be over-stepped when in follow-me?

It is also a possibility that I'm missing something else.   

### To Reproduce

1. Setup drone with a parameter-set (For example I used my default settings but with the SYS_VEHICLE_RESP 0.4, 0.3, 0.2, 0.1 and 0.01)
2. Place drone roughly 10-15m away from "Circle" follow-me mode will "follow" on a stationary target 
3. Drone switched on
4. Took off with drone to ~10m (Manual flight in position mode)
5. Enabled Follow me mode
6. Observe drone until it is stable
7. Return to launch
8. Re-do from 1 with next parameter you want to test OR re-test if you want more data on each parameter.

### Expected behavior

Expected the drone to be slower in reacting and less aggressive. But this was not noted on. 

### Flight Log

# Logs using SYS_VEHICLE_RESP = 0.01
https://review.px4.io/plot_app?log=8e772419-58ff-479d-b0a3-917e7b404613

# Logs using SYS_VEHICLE_RESP = 0.2
https://review.px4.io/plot_app?log=c1a8f7ad-fd37-40b5-8fad-06050be2ee3f

### Software Version

I use a slightly modified 1.14.0

### Flight controller

CUAV V5+

### Vehicle type

Multicopter

ripxorip, 2024-02-12:
I am interested in this behavior as well :eye:

Fraisa, 2024-02-13:
# smoothing: Could this help?

If anyone is familiar with the set-point handling. I've read the code in FlightTaskAutoFollowTarget.cpp and think I could have found something, would appreciate if someone took a glance.

There seems to be "Smoothening" between waypoints in normal-autonomous flights: See FlightTaskAuto.cpp: 
`PositionSmoothing::PositionSmoothingSetpoints smoothed_setpoints;
	_position_smoothing.generateSetpoints(
		_position,
		waypoints,
		_velocity_setpoint,
		_deltatime,
		force_zero_velocity_setpoint,
		smoothed_setpoints
	);
`

Would using this same function cause the drone to be "Smoother" in the case of follow-me? The are differences in how things are handled, but wanted to ask before testing anything as I'm new to this part of the code.

Still looking, but would love any feedback on this last find.