PX4/PX4-Autopilot #22664: [Bug] PX4 Boot overwrites all existing configuration on Septentrio GNSS receiver

SeptentrioGNSS, 2024-01-22:
### Describe the bug

The latest PX4 pro stable release 1.14.0 has a default config that overwrites all configurations of a Septentrio receiver (mosaic Go Heading - dual antenna option). This happens on every restart so the GNSS (external) receiver needs to be reconfigured after the boot sequence of the Pixhawk4 I am using. As this is a tedious task and requires more tool than sometimes at hand, is there an option to edit the auto configuration of streams on the GPS module as there is only a singular stream actually in use (see attachment) for the data transferred between the external GPS module and the PX4.  

All predefined configurations, even on other ports on the GPS module, are being removed/overwritten. 

### To Reproduce

The latest stable FW (1.14.0) is uploaded to a Holybro Pixhawk4 and the Septentrio receiver is connected via the GPS Module header with the following lanes connected:

The overwrite happens on every reboot of the PX4/Pixhawk4 so the GNSS receiver needs to be reconfigured after the reboot every time and only a singular COM1 output stream is actually in use. 

### Expected behavior

No problem with auto configuring a singular stream for the external GPS module to send data to the Pixhawk4. But it should not overwrite all existing config on the GNSS receiver **_and_** also should not have empty streams defined. A request to overwrite a singular stream is fine as the GNSS receiver can be configured in a way to stream on all the other ports if a specific configuration/data log is needed by the user that is kept and needed only to be set up once.

### Screenshot / Media

[image]
After boot - output stream on mosaic Go config

[image]
mosaic Go wired to Pixhawk4 (GPS Module 10pin)

[image]
Parameters > GPS configuration

### Flight Log

https://github.com/PX4/PX4-Autopilot/assets/151617650/7c624b40-fd94-4f50-85a3-7a7f5edab5fb
The predefined config of the external GNSS receiver is erased and the new config is send to the receiver after a boot sequence of the PX4.

no log available on this

### Software Version

PX4 Pro stable release 1.14.0 as QGC suggests when connecting the Pixhawk4 (Holybro) to the software tool with the Firmware Setup tab open. 

px4_fmu-v5_default 

### Flight controller

Holybro Pixhawk4

### Vehicle type

Rover

### How are the different components wired up (including port information)

External GNSS receiver (Septentrio mosaic Go Heading) connected to the GPS module with PWR, GND, RX, TX wired as the attachment above states

SeptentrioGNSS, 2024-01-23:
@AlexKlimaj To move this forward, is there any info you need from me? Or do I need to define a better scope of what would be workable as an end result?

AlexKlimaj, 2024-01-24:
Do you have anyone internally who can work on this? I know Seppe did the driver initially.

SeptentrioGNSS, 2024-01-26:
I am not available to work on this myself in the short and medium term so I am looking at the community or PX4 developers to help me in these fixes. I am however available to test changes made to confirm the fixes solve the problem and a pull request can be initiated. So any help would be greatly appreciated.

jonpeck, 2024-01-29:
Stream1 is only reset in the septentrio-gnss fork of the PX4-GNSSDrivers: [https://github.com/septentrio-gnss/Septentrio-PX4-GNSSDrivers/commit/1414c4c58c8ea755bca314fee5a8d34a7340294c](https://github.com/septentrio-gnss/Septentrio-PX4-GNSSDrivers/commit/1414c4c58c8ea755bca314fee5a8d34a7340294c)

AlexKlimaj, 2024-02-07:
It looks like the driver is creating many SBF streams on boot.

[image]

AlexKlimaj, 2024-02-08:
https://github.com/PX4/PX4-Autopilot/pull/22734

julianoes, 2024-03-03:
@AlexKlimaj should this be backported to v1.14.z?

SeptentrioGNSS, 2024-03-05:
> @AlexKlimaj should this be backported to v1.14.z?

If possible, yes this would be helpful to have in the first possible release