PX4/PX4-Autopilot #22683: [Bug] UAV Model fails to sapwn in Gazebo with a custom world

mzahana, 2024-01-25:
### Describe the bug

I am using PX4 `v1.14` and Gazebo `garden`. I am trying to simulate a quadcopter (e.g. `x500`) in a custom world. I plaved my custom world in `Tools/simulation/gz/worlds` directory. Then I  used the command
```bash
PX4_GZ_WORLD=mountains make px4_sitl gz_x500
```

Gazebo opens up with the custom world. However, the x500 quadcopter is not spawned. I get the following output on the terminal

```bash
px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4001
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 274 bytes, decoded 274 bytes (INT32:11, FLOAT:3)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
INFO  [init] starting gazebo with world: /home/user/shared_volume/PX4-Autopilot/Tools/simulation/gz/worlds/mountains.sdf
WARN  [init] PX4_GZ_MODEL_NAME or PX4_GZ_MODEL not set using PX4_SIM_MODEL.
INFO  [gz_bridge] world: mountains, model name: x500_0, simulation model: x500
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-user'
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
ERROR [gz_bridge] Service call timed out
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start
ERROR [px4] Startup script returned with return value: 256

```

**NOTE** I am running this setup inside a docker container.

The simulation works fine with the default world. The interesting thing is if I copy the content of my custom world into the world file named `default.sdf` and run the simulation with the `x500` model and the `default.sdf`,  the custom world works fine ! However, if I use any different world with a file name different than `default.sdf`, the world shows up in Gazebo, but the drone model fails to spawn!

### To Reproduce

* Clone the PX4-Autopilot repo and check out `v1.14`
* Place a custom world model `custom.sdf` in `PX4-Autopilot/Tools/simulation/gz/worlds/`
* run the simulation using `PX4_GZ_WORLD=custom make px4_sitl gz_x500`

### Expected behavior

Gazebo runs with the custom world with the `x500` model in it.

### Flight Log

No logs as  PX4 SITL fails to start

### Software Version

All running inside a docker container
* Ubuntu 22
* PX4  `v1.14`
* Gazebo `garden`

### Flight controller

SITL

### Vehicle type

Multicopter

mzahana, 2024-01-25:
@Jaeyoung-Lim any clue ?

dirksavage88, 2024-01-31:
@mzahana what does your custom world sdf look like? Are you including the following plugins? 
 ```
<physics type="ode">
      <max_step_size>0.004</max_step_size>
      <real_time_factor>1.0</real_time_factor>
      <real_time_update_rate>250</real_time_update_rate>
    </physics>
    <plugin name='gz::sim::systems::Physics' filename='gz-sim-physics-system'/>
    <plugin name='gz::sim::systems::UserCommands' filename='gz-sim-user-commands-system'/>
    <plugin name='gz::sim::systems::SceneBroadcaster' filename='gz-sim-scene-broadcaster-system'/>
    <plugin name='gz::sim::systems::Contact' filename='gz-sim-contact-system'/>
    <plugin name='gz::sim::systems::Imu' filename='gz-sim-imu-system'/>
    <plugin name='gz::sim::systems::AirPressure' filename='gz-sim-air-pressure-system'/>
    <plugin name='gz::sim::systems::ApplyLinkWrench' filename='gz-sim-apply-link-wrench-system'/>
    <plugin name='gz::sim::systems::Sensors' filename='gz-sim-sensors-system'>
      <render_engine>ogre2</render_engine>
    </plugin>
    <gui fullscreen='false'>
      <plugin name='3D View' filename='GzScene3D'>
        <gz-gui>
          <title>3D View</title>
          <property type='bool' key='showTitleBar'>0</property>
          <property type='string' key='state'>docked</property>
        </gz-gui>
        <engine>ogre2</engine>
        <scene>scene</scene>
        <ambient_light>0.5984631152222222 0.5984631152222222 0.5984631152222222</ambient_light>
        <background_color>0.8984631152222222 0.8984631152222222 0.8984631152222222</background_color>
        <camera_pose>-6 0 6 0 0.5 0</camera_pose>
      </plugin>
      <plugin name='World control' filename='WorldControl'>
        <gz-gui>
          <title>World control</title>
          <property type='bool' key='showTitleBar'>0</property>
          <property type='bool' key='resizable'>0</property>
          <property type='double' key='height'>72</property>
          <property type='double' key='width'>121</property>
          <property type='double' key='z'>1</property>
          <property type='string' key='state'>floating</property>
          <anchors target='3D View'>
            <line own='left' target='left'/>
            <line own='bottom' target='bottom'/>
          </anchors>
        </gz-gui>
        <play_pause>1</play_pause>
        <step>1</step>
        <start_paused>1</start_paused>
      </plugin>
      <plugin name='World stats' filename='WorldStats'>
        <gz-gui>
          <title>World stats</title>
          <property type='bool' key='showTitleBar'>0</property>
          <property type='bool' key='resizable'>0</property>
          <property type='double' key='height'>110</property>
          <property type='double' key='width'>290</property>
          <property type='double' key='z'>1</property>
          <property type='string' key='state'>floating</property>
          <anchors target='3D View'>
            <line own='right' target='right'/>
            <line own='bottom' target='bottom'/>
          </anchors>
        </gz-gui>
        <sim_time>1</sim_time>
        <real_time>1</real_time>
        <real_time_factor>1</real_time_factor>
        <iterations>1</iterations>
      </plugin>
      <plugin name='Entity tree' filename='EntityTree'/>
    </gui>
    <gravity>0 0 -9.8</gravity>
    <magnetic_field>6e-06 2.3e-05 -4.2e-05</magnetic_field>
    <atmosphere type='adiabatic'/>
    <scene>
      <grid>false</grid>
      <ambient>0.4 0.4 0.4 1</ambient>
      <background>0.7 0.7 0.7 1</background>
      <shadows>true</shadows>
    </scene>
```
Also one thing I notice is a lot of the world files from the gazebo model repo don't include <world> tags on the sdf, so px4 sitl thinks it's a model and not a world.

mzahana, 2024-01-31:
@dirksavage88 yes. If the same exact world is saved in a file named `default.sdf` it works. If it is saved with any other name (same content), it does not work.

frede791, 2024-02-15:
Hey @mzahana I might know why this is not working: When you name your world, the world name needs to match with the name of the file. So when you call your world `mountains.sdf` your world must have the name `mountains`. I suspect that since it works for default, your world is still named default. So to fix this just adjust the top of your file to look like:

```shell
<sdf version='1.9'>
  <world name='mountains'>
...
```

mzahana, 2024-02-24:
@frede791 thanks !

How did I miss that!   😳
That solved it.

closed 2024-02-24 by mzahana

frede791, 2024-02-24:
It's not well established. I'll make a PR to better highlight this in the user guide. 
Glad I could be of help!