PX4/PX4-Autopilot #22684: [Bug] SPI DMA transfer is not locked with multiple devices

dmammolo, 2024-01-25:
### Describe the bug

On my own px4 fork I have observed weird issues with a pix32v6/pixhawk6c on the IMU driver side.
The issue only pops up in some edge cases, i.e. after I added a random line of code completely unrelated to the IMU drivers.

The issue is following:
- icm42688p: many error/overflows/resets
- bmi055: driver looks fine but the published accel/gyro data is wrong. Acceleration in Z jumps around from ~1g to 0. Yaw gyro has on average >0 rad/s when static and thus the heading starts rotating.

I have put quite some effort in debugging this issue and am fairly sure what causes the problem. For more background please read the issue I created here: https://github.com/apache/nuttx/issues/11594

### Solution:
The SPI transfer needs to be locked if multiple devices use the same SPI bus and/or an IMU such as BMI055 or BMI088, which require requesting data for gyro and accel separately. 

I have verified with release 1.13.3 and 1.14.0 that the SPI transer function is not locked (nor `LOCK_PREEMPTION` nor `LOCK_THREADS`):
https://github.com/PX4/PX4-Autopilot/blob/da28d9a7f2c9d664631c3ee536fa6f1c116d65de/src/lib/drivers/device/nuttx/SPI.cpp#L130-L163

IMO in this case it needs to be `LOCK_PREEMPTION`.

### To Reproduce

Don't have a code version I can share and it is hard to reproduce as I suspect some memory shifts and variations of modules/drivers might change the timing.

### Expected behavior

No errors in the driver and uncorrupted accel/gyro data. 

### Flight Log

-

### Software Version

1.13.3 and 1.14.0

### Flight controller

pix32v6, pixhawk 6C

### Vehicle type

None

davids5, 2024-01-25:
> I added a random line of code completely unrelated to the IMU drivers

Have you effected the priorities of the system or the stack usage?

dmammolo, 2024-01-25:
> Have you effected the priorities of the system or the stack usage?

I have some custom drivers and modules but they should not affect the SPI drivers.

davids5, 2024-01-26:
@dmammolo 

The last time I had an issues like this it was a piece of code in the FAT32 that was cleverly reusing a buffer to save the cost of a copy. It was reading the buffer prior to a DMA completion and corrupting it. 

How is DCACHE set in your defconfig settings WB or WT? 

I suspect you have altered the memory layout with the driver, and module you have added in such a way that there is some cache line overlap. Have you compared  your fork's memory map (nm -C) to that of main?

I would start there rather then change something that covers up the root cause of a new symptom, especially given the fact that this code pattern has been used since the F7 was first used (in the FMUV5 production since 2017)

dmammolo, 2024-01-26:
> The last time I had an issues like this it was a piece of code in the FAT32 that was cleverly reusing a buffer to save the cost of a copy. It was reading the buffer prior to a DMA completion and corrupting it.

interesting, but from the implementation in stm32_spi.c the buffer reading happens after DMA completion using memcpy.

> 
> How is DCACHE set in your defconfig settings WB or WT?

The same as the upstream defconfig -> WB.

> 
> I suspect you have altered the memory layout with the driver, and module you have added in such a way that there is some cache line overlap. Have you compared your fork's memory map (nm -C) to that of main?
I did now and it looks similar, i.e. I could not find a driver/module buffer that could overlap with g_spi1_rxbuf. What/how would you check for cache line overlaps yourself?

> 
> I would start there rather then change something that covers up the root cause of a new symptom, especially given the fact that this code pattern has been used since the F7 was first used (in the FMUV5 production since 2017)

that's not correct, the `spi_exchange()` logic of stm32_spi.c for the F7 differs to the H7. For F7 the cache invalidation happens after the DMA completion and before reading the rx buffer (memcpy)

davids5, 2024-01-26:
> that's not correct, the `spi_exchange()` logic of stm32_spi.c for the F7 differs to the H7. For F7 the cache invalidation happens after the DMA completion and before reading the rx buffer (memcpy)

I stand corrected. F7 does do the invalidate once and after the DMA completion.  

Still the same questions: What is touching the buffer space and reading in the cache lines? Why does it only fail on your fork?

dmammolo, 2024-01-26:
From the post I shared I understand that during the DMA operation there is a chance for `speculative reads`. These apparently can happen independent of any data read instructions...(extracted from [here](https://community.arm.com/support-forums/f/architectures-and-processors-forum/52740/on-cortex-m7-can-speculative-access-bring-accessed-data-to-d-cache))

Clearly this is very hard to proof if that's the case in my fork.

Sorry my previous comment was not formatted well and you missed my comment regarding the memory map. I.e. how would you verify if there is a cache line overlap from somewhere else using the memory map?

davids5, 2024-01-26:
@dmammolo What is below (before) and above (after)  `g_spi1_rxbuf` in both memory maps?

dmammolo, 2024-01-29:
I do not see anything that looks weird, this is what I could (manually) extract that is close to `g_spi1_rxbuf`. For my fork:
```
240119cc B param_values
240119d0 b g_dev
24011a28 b once.0
24011a40 b g_spi1_rxbuf
24011e40 b g_spi1_txbuf
```

upstream v1.14.0:
```
2400c49c b vbus_present_prev
2400c4b0 b g_bbsram
2400c500 b once.0
2400c520 b g_spi1_rxbuf
2400c920 b g_spi1_txbuf
```

davids5, 2024-01-29:
> I do not see anything that looks weird, this is what I could (manually) extract that is close to `g_spi1_rxbuf`. For my fork:
>
> ```
> 240119cc B param_values
> 240119d0 b g_dev
> 24011a28 b once.0
> 24011a40 [...]

The only obvious change is a shift of the addresses by 21792 bytes upward. Could is now be getting accessed by stack smashing? 

Can you reorg some of your static data and see if the problem goes away?

dmammolo, 2024-01-30:
Are there any tools you know that could help find possible stack smashing effects which points to the source?
As mentioned already, these issues start exactly with reorganization/adaptation of static data. I.e. I observed the issue to pop up after I changed the value if a `static constexpr int`. So yes reorganizing will probably on some point lead to a resolution of the effects but clearly it's not the solution.

davids5, 2024-01-30:
> Are there any tools you know that could help find possible stack smashing effects which points to the source? 

You can use [ARMv7-M Run Time Stack Checking](https://cwiki.apache.org/confluence/display/NUTTX/ARMv7-M+Run+Time+Stack+Checking) just keep in mind it is overly conservative. 

Agreed the reorg is not a solution, but the new layout has pointed to problem that should not be ignored.

sdirkwinkel, 2026-02-12:
@davids5  @dmammolo Was there any progress on this or is/was there a similar issue?

I had a copter fall out of the sky with v1.16.0 (locally built) and fmu-v6c
Both IMUs on SPI1 failed within 1ms from the log. I've not been able to reproduce this (have tried scheduling imu reads as quickly as possible and running for a couple of hours)

Might also have been hardware failure with the 3.3v? Would anything else be affected by that (other things continued as normal in the log). After the next boot everything was fine again.

Do you have an idea of what I could try to reproduce / look at?