PX4/PX4-Autopilot #22687: Multiple vehicle spawn error in ROS2 Humble

eogks8921, 2024-01-26:
### Describe the bug

Hello everyone,

I’m trying to spawn muliple iris_depth_camera vehicle in SITL
I followed below document.

[docs.px4.io](https://docs.px4.io/main/en/sim_gazebo_classic/multi_vehicle_simulation.html)
[Multi-Vehicle Simulation with Gazebo Classic | PX4 User Guide (main)](https://docs.px4.io/main/en/sim_gazebo_classic/multi_vehicle_simulation.html)
PX4 is the Professional Autopilot. Developed by world-class developers from industry and academia, and supported by an active world wide community, it powers all kinds of vehicles from racing and cargo drones through to ground vehicles and...

When I tried to build that, I met the error

[ERROR] [1705243878.281362595] [gazebo_ros_node]: Found multiple nodes with same name: /camera_controller. This might be due to multiple plugins using the same name. Try changing one of the the plugin names or use a different ROS namespace. This error might also result from a custom plugin inheriting from another gazebo_ros plugin and the custom plugin trying to access the ROS node object hence creating multiple nodes with same name. To solve this try providing the optional node_name argument in gazebo_ros::Node::Get() function.

software versions are below.

gazebo version : classic
ROS2 version : ROS2 humble
PX4 version : 1.14.0 stable

Please give me any advice or ideas to solve the problem.
Thank you.

### To Reproduce

I follow doc.

https://docs.px4.io/main/en/sim_gazebo_classic/multi_vehicle_simulation.html

### Expected behavior

Build Error

### Screenshot / Media

None

### Flight Log

None

### Software Version

PX4 : 1.14.0
ROS2 : Humble
Simulation : Gazebo classic

### Flight controller

Simulation(iris_depth_camera)

### Vehicle type

Multicopter

eogks8921, 2024-02-08:
I solved problem.

Booooms, 2024-10-12:
How did you solve it? I keep getting errors as shown below.

[ERROR] [1728756003.612131841] [gazebo_ros_node]: Found multiple nodes with same name: /imu_plugin. This might be due to multiple plugins using the same name. Try changing one of the the plugin names or use a different ROS namespace. This error might also result from a custom plugin inheriting from another gazebo_ros plugin and the custom plugin trying to access the ROS node object hence creating multiple nodes with same name. To solve this try providing the optional node_name argument in gazebo_ros::Node::Get() function.