PX4/PX4-Autopilot #22695: [Bug] About compiler issue

JustinHsuNctu, 2024-01-29:
### Describe the bug

Hello! Recently, I want to use PX4-Autopilot to control the UAV in GAZEBO simulator. I tried to compile the repo which version is V1.14.0. when I executed "make px4_sitl," I got an error about protobuf version conflict.  The error messages are in "Additional context."
I tried to down grade protobuf to 2.6.1. But the error is still there. 
Any one could help me about it?

### To Reproduce

The environment that I use is windows subsystem linux, WSL2. The process that I follow are:
1. cd ~
2. git clone https://github.com/PX4/PX4-Autopilot.git --recursive
3. bash ./PX4-Autopilot/Tools/setup/ubuntu.sh
4. exit WSL and shutdown WSL
5. reboot wsl
6. cd PX4-Autopilot/
7.  make px4_sitl

### Expected behavior

Pass the compile and execute GAZEBO

### Flight Log

NO.

### Software Version

V1.14.0

### Flight controller

NO

### Vehicle type

Multicopter

### Additional context

ubuntu@DESKTOP-129CA14:~/PX4-Autopilot$ make px4_sitl
-- PX4 version: v1.15.0-alpha1-457-gf40ede6087 (1.15.0)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3")
-- PX4 config file: /home/ubuntu/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- ROOTFSDIR .
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 11.4.0
-- The C compiler identification is GNU 11.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
/usr/bin/python3: Error while finding module specification for 'symforce.symbolic' (ModuleNotFoundError: No module named 'symforce')
-- Could NOT find gz-transport (missing: gz-transport_DIR)
-- Looking for gz-transport12 -- found version 12.2.1
-- Searching for dependencies of gz-transport12
[... 53 lines cut ...]
/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"gz_bridge\" -DPROTOBUF_USE_DLLS -DPX4_MAIN=gz_bridge_app_main -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I/home/ubuntu/PX4-Autopilot/boards/px4/sitl/src -I/home/ubuntu/PX4-Autopilot/platforms/posix/src/px4/common/include -I/home/ubuntu/PX4-Autopilot/build/px4_sitl_default -I/home/ubuntu/PX4-Autopilot/build/px4_sitl_default/src/lib -I/home/ubuntu/PX4-Autopilot/platforms/posix/src/px4/generic/generic/include -I/home/ubuntu/PX4-Autopilot/platforms/common -I/home/ubuntu/PX4-Autopilot/platforms/common/include -I/home/ubuntu/PX4-Autopilot/src -I/home/ubuntu/PX4-Autopilot/src/include -I/home/ubuntu/PX4-Autopilot/src/lib -I/home/ubuntu/PX4-Autopilot/src/lib/matrix -I/home/ubuntu/PX4-Autopilot/src/modules -I/home/ubuntu/PX4-Autopilot/platforms/posix/include -I/home/ubuntu/PX4-Autopilot/build/px4_sitl_default/external/Install/include -isystem /usr/include/gz/transport12 -isystem /mnt/c/ProgramData/anaconda3/Library/include -isystem /usr/include/gz/utils2 -isystem /usr/include/gz/msgs9 -isystem /usr/include/gz/math7 -isystem /usr/include/uuid -O2 -g -DNDEBUG -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -Wreorder -Wno-overloaded-virtual -frtti -fexceptions -O2 -I/usr/include/uuid -std=gnu++17 -MD -MT src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceWheel.cpp.o -MF src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceWheel.cpp.o.d -o src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceWheel.cpp.o -c /home/ubuntu/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceWheel.cpp
In file included from /usr/include/gz/msgs9/gz/msgs/pointcloud_packed.pb.h:17,
                 from /usr/include/gz/msgs9/gz/msgs/detail/PointCloudPackedUtils.hh:24,
                 from /usr/include/gz/msgs9/gz/msgs.hh:23,
                 from /home/ubuntu/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceWheel.hpp:38,
                 from /home/ubuntu/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceWheel.cpp:34:
/usr/include/gz/msgs9/gz/msgs/details/pointcloud_packed.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
compilation terminated due to -Wfatal-errors.
[683/967] Building CXX object src/modules/simulation/gz_br...es__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o
FAILED: src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o
/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"gz_bridge\" -DPROTOBUF_USE_DLLS -DPX4_MAIN=gz_bridge_app_main -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I/home/ubuntu/PX4-Autopilot/boards/px4/sitl/src -I/home/ubuntu/PX4-Autopilot/platforms/posix/src/px4/common/include -I/home/ubuntu/PX4-Autopilot/build/px4_sitl_default -I/home/ubuntu/PX4-Autopilot/build/px4_sitl_default/src/lib -I/home/ubuntu/PX4-Autopilot/platforms/posix/src/px4/generic/generic/include -I/home/ubuntu/PX4-Autopilot/platforms/common -I/home/ubuntu/PX4-Autopilot/platforms/common/include -I/home/ubuntu/PX4-Autopilot/src -I/home/ubuntu/PX4-Autopilot/src/include -I/home/ubuntu/PX4-Autopilot/src/lib -I/home/ubuntu/PX4-Autopilot/src/lib/matrix -I/home/ubuntu/PX4-Autopilot/src/modules -I/home/ubuntu/PX4-Autopilot/platforms/posix/include -I/home/ubuntu/PX4-Autopilot/build/px4_sitl_default/external/Install/include -isystem /usr/include/gz/transport12 -isystem /mnt/c/ProgramData/anaconda3/Library/include -isystem /usr/include/gz/utils2 -isystem /usr/include/gz/msgs9 -isystem /usr/include/gz/math7 -isystem /usr/include/uuid -O2 -g -DNDEBUG -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -Wreorder -Wno-overloaded-virtual -frtti -fexceptions -O2 -I/usr/include/uuid -std=gnu++17 -MD -MT src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o -MF src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o.d -o src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o -c /home/ubuntu/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceServo.cpp
In file included from /usr/include/gz/msgs9/gz/msgs/discovery.pb.h:17,
                 from /usr/include/gz/transport12/gz/transport/Discovery.hh:59,
                 from /usr/include/gz/transport12/gz/transport.hh:26,
[... cut ...]
                 from /home/ubuntu/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceServo.cpp:34:
/usr/include/gz/msgs9/gz/msgs/details/discovery.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
compilation terminated due to -Wfatal-errors.
[684/967] Building CXX object src/modules/replay/CMakeFiles/modules__replay.dir/Replay.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1

JustinHsuNctu, 2024-01-29:
I found the solution! In WSL2, all commands should execute in Windows PowerShell environment.
It is described on PX4 official website.
"All operations to install and build PX4 must be done within a WSL Shell (you can use the same shell that was used to install WSL2 or open a new one)."
from "https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html"

Igor-Misic, 2024-01-31:
Thanks for sharing the solution!

closed 2024-01-31 by Igor-Misic

xianxian437, 2024-02-01:
> I found the solution! In WSL2, all commands should execute in Windows PowerShell environment. It is described on PX4 official website. "All operations to install and build PX4 must be done within a [...]

I have encountered the same problem, but I cannot solve it through your method. Can you provide more detailed suggestions?

JiazeCai, 2024-06-15:
I am currently still getting the exact same issue. Is this somehow fixed?

JiazeCai, 2024-06-15:
Yeah, I turns out to install Ubuntu 22.04 system to get everything work out. Thanks for your suggestion!

Strevino100, 2024-07-13:
I have ubuntu 22.04 and still have this error. Any ideas?

maik-basso-isi, 2024-08-06:
The same happens here. Is there any workaround to solve the problem?