PX4/PX4-Autopilot #22707: [Bug] Matek H743-slim-v3 PWM outputs S3, S4, S5, S6 don't work

svpcom, 2024-01-31:
### Describe the bug

In px4 (both release 1.14.1 and latest master) outputs S3, S4, S5, S6 (PA0, PA1, PA2, PA2 attached to TIM5) don't output any signals (tested with oscilloscope) both in PWM and DSHOT modes.  The same hardware with Ardupilot works fine.

### To Reproduce

1. make matek_h743-slim upload
2. In QGC select copter profile and assign motors to outputs S1, S2, S3, S4
3. Attach ESCs (or oscilloscope probes) to these outputs
4. Motors on S1 and S2 will beep, but S3 and S4 will be silent

### Expected behavior

Expected that S3, S4, S5, S6 will work.  Other outputs (S1, S2, S7 - S12) works normally. It seems that something bad with DMA and/or timer mapping.

### Flight Log

No flight log available

### Software Version

px4 v1.14.1 or latest main

### Flight controller

Matek H743-slim-v3

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Motors wired to outputs S1, S2, S3, S4

IgorSheyko, 2024-02-03:
the same issue and ad-hoc solution is described here https://www.youtube.com/watch?v=6uVFTe3Kmlk

+1 waiting for a long-term solution

svpcom, 2024-02-03:
@IgorSheyko I didn't found solution in this video except of to ignore S3-S6 outputs.

Woojin-Crive, 2024-04-01:
disable dronecan!

tonycake, 2025-02-12:
This isn't only a Matek problem, seems that any H743 based FC has the issue. 
Looking at the debugger, CCMR registers are not being set on Tim5, even though other registers seem to be ok (CCR, CNT, etc). 
Disabling DRONECAN, as suggested above, does provide a temporary workaround.

julianoes, 2025-11-28:
DroneCAN needs to be on a different timer, using the override `CONFIG_BOARD_UAVCAN_TIMER_OVERRIDE`.

E.g. https://github.com/PX4/PX4-Autopilot/blob/0618b0b529d689609cebbcee500ee14d83ae58fe/boards/ark/fmu-v6x/default.px4board#L49