PX4/PX4-Autopilot #22709: PX4_SITL position mode height drift [Bug]

dirksavage88, 2024-01-31:
### Describe the bug

When flying x500/px4vision in px4_sitl the quadcopter will not hover stable and will drift higher and higher (infinitely) unless the throttle is lowered. This issue was repeated with both a ps4 dualshock and logitech joystick. The issue did not occur in hold mode. 

### To Reproduce

1) Start a px4_sitl x500 or px4vision instance
2) connect joystick and make sure qgc shows the joystick page responds to stick input
3) set to position mode, take off and attempt to hover. leave throttle at approximate thrust to maintain stable hover
4) observe height drift upward infinitely unless throttle is lowered
5) switch to hold mode and observe height drift increase not an issue

### Expected behavior

PX4 SITL quadcopters in position mode should not drift upward infintiely

### Flight Log

N/A

### Software Version

Main

### Flight controller

POSIX

### Vehicle type

Multicopter

MaEtUgR, 2024-02-06:
Would it be possible to get a log from such a simulated flight?
In SITL logs are stored under `PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/log/2024-xx-xx/xx_xx_xx.ulg`

dirksavage88, 2024-02-08:
Here is the log that demonstrates it. If I look at the joystick page in QGC, I can see the stick for throttle slowly increase without touching any of the buttons. If I try to level out it will slowly increase again. 
[02_00_15.zip](https://github.com/PX4/PX4-Autopilot/files/14203508/02_00_15.zip)

MaEtUgR, 2024-02-12:
@dirksavage88 Thanks for the log. Next time you can directly upload it to [PX4 Flight Review](https://docs.px4.io/main/en/log/flight_log_analysis.html#flight-review-online-tool):
https://review.px4.io/plot_app?log=c5532491-2421-4af7-a2c6-205f357ade66

The data in the log shows that the autopilot listens to the stick input like expected. But the way the stick input behaves leaves me puzzled.
[image]

It's coming through MAVLink e.g. from the ground station and it seems to drift upwards presumably every time you let go of the stick. What kind of Joystick are you using? I presume a physical joystick. Could it have physical or electronic issues in the joystick gimbals or a driver issue in the ground station OS? Could you compare against behavior with the [QGC Virtual Joystick](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/settings_view/virtual_joystick.html)? Could you check the proper functionality of the physical joystick in the OS e.g.
Windows: https://www.howtogeek.com/241421/how-to-calibrate-your-gaming-controller-in-windows-10/
Linux: https://unix.stackexchange.com/a/80935

dirksavage88, 2024-02-12:
The height drift upwards does not occur with only the qgc virtual joysticks.

MaEtUgR, 2024-02-12:
Just tested with a Logitech F310 and latest QGC daily 
[image]
on Windows 11 against SITL and it works like expected 🤔
Did you calibrate the joystick (3rd tab under Joystick)?
[image]

dirksavage88, 2024-02-12:
Try enabling circle correction & deadbands

dirksavage88, 2024-02-12:
also the drift is more pronounced when you are at 50% throttle