PX4/PX4-Autopilot #22714: [Bug] Very large trajectory_setpoint.accelerations (horizontal) when MPC_MAN_TILT_MAX is big (e.g. 90 degrees) in ALTITUDE mode

alistair-blueflite, 2024-02-01:
### Describe the bug

When MPC_MAN_TILT_MAX is set very high (e.g. 90 degrees) and the vehicle is in ALTITUDE mode, then the resulting horizontal accelerations from the trajectory_setpoint become very high with horizontal stick input (e.g. a 10% pitch stick input results in horizontal accelerations around 10million when MPC_MAN_TILT_MAX = 90)

@julianoes 
@MaEtUgR 
@bkueng 

### To Reproduce

set MPC_MAN_TILT_MAX very high (> 70). Put the drone in ALTITUDE mode and observe the trajectory_setpoint.accelerations in the console in QGC as you move the pitch and roll sticks around. You will likely notice that even with very small stick input (10%), the output accelerations become very large

### Expected behavior

When setting high MPC_MAN_TILT_MAX, the resulting trajectory_setpoint.accelerations should be within reason (not 10million).

This is suspected to be a result of using tan() in 'generateAccelerationSetpoints()' in StickTIltXY.cpp in the flight_mode_manager module. Curious as to why use something like tan() over sine() or a linear mapping to scale the sticks, since tan() has the issue of blowing up to very large values. 

### Flight Log

N/A

### Software Version

modified version of v1.14-beta2

### Flight controller

CubeOrange

### Vehicle type

None

mcsauder, 2024-02-01:
With any component of vertical thrust a 90 degree bank angle might be possible. This is not possible for a multicopter.

It is illogical to expand the max bank angle to 90 degrees and expect to maintain altitude.

This is, however, possible of fixed wing aircraft capable of generating some component of vertical thrust via yaw angle.

MaEtUgR, 2024-02-15:
Thanks for the report. I assume you're completely right. No one ever went that far. Let me see what a good solution is.

> Curious as to why use something like tan() over sine() or a linear mapping to scale the sticks

The idea is to have the stick input represent a tilt angle and not cause a big tilt change around the center of the stick and almost none further away from the center. 👀

MaEtUgR, 2024-02-15:
I can reproduce the calculation must be completely incorrect in case `MPC_MAN_TILT_MAX` is set to 90° even for what it was intended for. I would have expected it to only blow up once you get very close to commanding 90°.
[image]

The calculation is here: https://github.com/PX4/PX4-Autopilot/blob/cb09dde606861b90e38b1682eebd3fd91be17ab7/src/modules/flight_mode_manager/tasks/Utility/StickTiltXY.cpp#L52

Probably I'll add a unit test.

MaEtUgR, 2024-02-16:
My proposed fix: https://github.com/PX4/PX4-Autopilot/pull/22769

PR #22769 (2024-02-20): StickTiltXY: Fix too high maximum tilt problem
### Solved Problem
Reported by @alistair-blueflite when `MPC_MAN_TILT_MAX` is set to close to or 90° the accelerations go insane and I even had it that the stick input reverses direction at 90°.

Fixes #22714

### Solution
- Consider 1° - 70° the recommended useful range. This corresponds to a 3:1 thrust-to-weight ratio.
- If the pilot sets `MPC_MAN_TILT_MAX` higher, guard against the calculations leading to a hardly steerable drone.
Note: In stabilized mode `MPC_MAN_TILT_MAX` can be up to 180° and the controller will track it.
- Add unit tests.

 [... cut]
closed 2024-02-20 by MaEtUgR